letscontrolit/ESPEasy #326: Documentation of Plugins

Grovkillen, 2017-05-31:
I just started this page: https://www.letscontrolit.com/wiki/index.php/Device_Plugins

We need to start documenting each (official) plugin and link to their pages:
https://www.letscontrolit.com/wiki/index.php/Plugin<plugin_number>

On the specific page we need to link to devices/sensors etc. that might be needed to use the plugin and or other explanations.

krikk, 2017-05-31:
don't understand why another list, if we have already a [list of supported devices](https://www.letscontrolit.com/wiki/index.php/Devices), because 2 lists, more work to keep it updated, and developers are lazy... so no redundant info...

krikk, 2017-05-31:
wouldn't it be better to add another column to the supported devices list?

Grovkillen, 2017-05-31:
Please elaborate on the "one list fits all". My thought is that since multiple devices/sensors/actuators can work with one or more plugins the matrix table will be hard to read. The list I just made would be the go-to list for the plugin information (pure info, only links to the devices that is confirmed to work). How they are installed and how to setup the plugin belong to the device page.

Just my thought... :)

Grovkillen, 2017-05-31:
I will populate the list I started and see if it is unnecessary. I can easily migrate it to the other list later. (I do all big tables in Notepad++ so it is no big deal to fix.)

Thanks for the input.

psy0rz, 2017-05-31:
@Grovkillen https://www.letscontrolit.com/wiki/index.php/ESPEasy#Hardware_page is gone?

That used to point to the "hardware page" of the ESPEasy webinterface. (the place where you configure GPIO bootstates and wifi status leds)

Grovkillen, 2017-06-01:
https://www.letscontrolit.com/wiki/index.php/ESP_Easy_web_interface#Hardware_page

It is moved to the sub page of "ESP web interface".

krikk, 2017-06-01:
not good, we should not break links that are in the firmware! 
[http://www.letscontrolit.com/wiki/index.php/ESPEasy#Hardware_page](http://www.letscontrolit.com/wiki/index.php/ESPEasy#Hardware_page) is linked from hardware page, can we do a redirect??

Grovkillen, 2017-06-01:
Well we needed to clean up the main page and since "Hardware page" was a section of that main page a redirect is not possible (as far as I know).

Pages can be redirected by setting this code
"#REDIRECT [[pagename]]"
on the old page....

Grovkillen, 2017-06-01:
This is the best I can do.... :/ Sorry for any inconvenience but it was a much needed clean-up of the main page.
https://www.letscontrolit.com/wiki/index.php/ESPEasy#Hardware_page

krikk, 2017-06-01:
ok, then i will correct the link in the source code

Grovkillen, 2017-06-01:
Okay, I now have a new idea. I will hear your argument about not using a new list and remove it. But since we need to be better to link to the wiki from the firmware i suggest that we start using the approach with:

https://www.letscontrolit.com/wiki/index.php/Plugin<plugin_number>

from within the web gui of ESP Easy. The plugin number can then be automatically fetched from the plugin number within the code (but without the prefix 0 i guess...). But we will not add ANY information on those pages except a redirect to whatever page is most suitable for the plugin. Please see how I redirected this page.

Hardcoded no redirect.
https://www.letscontrolit.com/wiki/index.php?title=Plugin52&redirect=no

The link https://www.letscontrolit.com/wiki/index.php/Plugin52 just bounce you to https://www.letscontrolit.com/wiki/index.php/S8 which is the best unit to describe the plugin....

What do you think?

krikk, 2017-06-01:
we already have this links to this plugin pages, this is even a feature in R120 :)

krikk, 2017-06-01:
see here: https://github.com/letscontrolit/ESPEasy/blob/mega/src/WebServer.ino#L1432

Grovkillen, 2017-06-01:
Sorry, didn't realize that! Then I should only care about the redirects working then. :)

Grovkillen, 2017-06-01:
Please close this issue.

Grovkillen, 2017-06-01:
We can then close this issue since I thought it was needed to have a cross reference from the web interface and the wiki. I now see that the redirect from plugin-number-page is sufficient!

Grovkillen, 2017-06-01:
Shouldn't we explain the indicators more in detail on each plugin? I just made this example for the "sysinfo" plugin:

https://www.letscontrolit.com/wiki/index.php/SysInfo

krikk, 2017-06-01:
good idea 👍

psy0rz, 2017-06-01:
indeed nice!

closed 2023-09-03 by TD-er