letscontrolit/ESPEasy #336: Displays: No setting for refresh rate anymore

thymjan, 2017-06-07:
### Steps to reproduce
Connect a Display

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
Tell us what should happen?
The setting for the refresh rate disappeared. The default refresh rate seems to be 60 sec.
In some environments its important to refresh the display more frequently

### Actual behavior
Tell us what happens instead?
Setting for the refresh rate is not possible

### System configuration
Hardware:
adafruit huzzah
Software or git version:
git version dated 2017-0607

aero83, 2017-06-08:
I have the same issue,  wemos D1 mini.  ssd1306 Oled display.  2.0r10 dev firmware.
 It appears to be a 60 sec update rate with no ability to change it.

psy0rz, 2017-06-10:
krikk fixed it. if it still doesnt work in the new version let us know.

closed 2017-06-10 by psy0rz