letscontrolit/ESPEasy #347: Rules engine force lower case in current code?

Grovkillen, 2017-06-09:
### Steps to reproduce
How can we trigger this problem?

Create a simple rule:

```
On System#Boot do   
 Publish %sysname%/TEST,TeSt
  timerSet,1,30
endon
On Rules#Timer=1 do
 Publish %sysname%/TEST,TeSt
  timerSet,1,30
endon
```

### Expected behavior
I'd expect the MQTT to be published as:

```
SysName/TEST
TeSt
```

### Actual behavior
It is published as:

```
sysname/test
test
```

mvdbro, 2017-06-10:
Seems to work fine here, tested with the same rule and listening with MQTTfx tool:
[image]

Grovkillen, 2017-06-10:
I have two units (one is mine, and one is my friends)... both compiled the other day, both lower case.

May someone else try the latest code, compile and test send through rules Publish command?

mvdbro, 2017-06-10:
Has there been a change to the rules engine? I'm using the 27-05-2017 commit from github.

Grovkillen, 2017-06-10:
Mine was self compiled some 3-5 days ago. I just thought it might be something that has been tinkered with within this week or so. Dev10 works as intended. This is after that.

mvdbro, 2017-06-10:
Just uploaded todays github version, seems also ok. Using OpenHab controller.

Grovkillen, 2017-06-10:
[image]
[image]

Grovkillen, 2017-06-10:
The "test" comes from the second rule sniff#cmdMQTT... forgot to circle that rule as well.

Both topic and payload is lower case for the rules publish, not the publish from devices.

mvdbro, 2017-06-10:
I can see that it flaws in your environment, These are my results using github version from june 10th:

[image]

And reboot situation
[image]

So both publishes are as expected. What could be different between our setups?

How does it show up in your log? This is how is looks on mine:
405782 : EVENT: Rules#Timer=1
405792 : ACT : Publish U18 Wemos Nexxt/TEST,TeSt
405833 : ACT : timerSet,1,30

Grovkillen, 2017-06-10:
```
3854040 : IMPT : [sniff#cmdMQTT] : 1.00
3854040 : EVENT: sniff#cmdMQTT=1.00
3854050 : ACT : Publish Portable_201/IP,192.168.0.157 TeSt
```

Log shows not lower case!?

psy0rz, 2017-06-25:
so this isn't a issue?

Grovkillen, 2017-06-25:
I will report back after next Dev release. The problem still exist on my unit. Self compiled though..?

psy0rz, 2017-06-25:
ok

Grovkillen, 2017-07-10:
I just compiled a version based on todays code. The problem is gone.

Grovkillen, 2017-09-11:
Great finding. @psy0rz could this be implemented? More are having issues with Rules+Publish.

psy0rz, 2017-09-14:
ahh that explains it, some plugin_write function somewhere changes the action string.

we should probably check all the PLUGIN_WRITE handlers and see which one is the culprit.

mvdbro, 2017-09-23:
These are the ones that break stuff. They act on a pass by ref variable:
_P054_DMX512.ino:        string.toLowerCase();
_P057_HT16K33_LED.ino:        string.toLowerCase();
_P065_DRF0299_MP3.ino:        string.toLowerCase();
_P067_HX711_Load_Cell.ino:        string.toLowerCase();
_P070_NeoPixel_Clock.ino:        string.toLowerCase();

mvdbro, 2017-09-23:
and more issues here:
_P054_DMX512.ino:          string.replace("  ", " ");
_P054_DMX512.ino:          string.replace(" =", "=");
_P054_DMX512.ino:          string.replace("= ", "=");
_P001_Switch.ino:            string.replace("-","#");
_P057_HT16K33_LED.ino:          string.replace("  ", " ");
_P057_HT16K33_LED.ino:          string.replace(" =", "=");
_P057_HT16K33_LED.ino:          string.replace("= ", "=");

psy0rz, 2017-09-23:
cool thanks

mvdbro, 2017-11-05:
This only affects the system when using TESTING status plugins. The respective plugin devs should have a look at this. If does not stop us from releasing V2.0 (that should only include stable plugins)

psy0rz, 2017-11-16:
@mvdbro wouldn't it we better if we make the Plugin_ptr string a const, to prevent these kind of mistakes?

i cant imagine a valid reason to allow a plugin to change this string?

psy0rz, 2017-11-16:
oh right, just released how its used in PLUGIN_GET_DEVICENAME

psy0rz, 2017-11-16:
@mvdbro although its indeed not a v2.0 issue, i fixed it anyway in all the plugins, to prevent more long debugging threads and fuzzy problems.

commit 7a2d1143 (2017-11-16): fix illegal string manipulations in PLUGIN_WRITE for some plugins. fixes #347 and #460
closed 2017-11-16 by psy0rz