letscontrolit/ESPEasy #350: P021 Level control

poedk, 2017-06-10:
### Steps to reproduce
Send Level control output to controller 

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Persistent

### Expected behavior
Output status should be sent every "delay"  

### Actual behavior
Output status is only sent when changed

### System configuration

Hardware:
D1 mini
Software or git version:
dev10

mvdbro, 2017-06-10:
Hmm, this device acts as a (local) switch. I would expect it's actual behavior...

poedk, 2017-06-10:
I use it to control my kegerator (looking forward to the inverse logic function).
My idea is that if I set delay to 60s, and status is posted every cycle, I could make a nice bar graph below my temperature (logged every 60s) graph in Grafana, showing when cooling is active. If i summarize the output it could tell me how many minutes the cooling has been active.

psy0rz, 2017-06-12:
this is a feature not a bug

closed 2017-06-12 by psy0rz