letscontrolit/ESPEasy #354: Add a PLUGIN_STOP function

psy0rz, 2017-06-12:
When PLUGIN_INIT is called, things like this might happen:
```Plugin_052_SoftSerial = new SoftwareSerial(...);```

We need a way to deallocate that memory when the user is switching around plugins. This will save memory, but also make issue-reporting more consistent: Currently some people might run into memory issues when switching between certain plugins. These issues will be vague and hard to reproduce. If we deallocate memory we can solve this.

Maybe some hardware also needs to be in a certain state before switching to another plugin. This also can be done in the PLUGIN_STOP function.

psy0rz, 2017-06-12:
perhaps PLUGIN_EXIT is a better name now that i think of it

mvdbro, 2017-06-17:
I agree that this can/should be improved, but do we really want to have this in 2.0.0. milestone?

psy0rz, 2017-06-19:
yeah you're right. there is a probability we will break new stuff when we implement this ;) in the long run it will solve problems.

psy0rz, 2018-01-16:
this is already done and is called PLUGIN_EXIT and is in v2.0, and has issues when someone disables a plugin, while the plugin uses delayBackground or backgroundprocessing().

Unfortunately we didn't had a separate v2.0 branch back then.

closed 2018-01-16 by psy0rz