letscontrolit/ESPEasy #355: Conctant watchdog reset: Nodemcu - when Default pin states 0,9,14 are low

SloMusti, 2017-06-13:
The issue occurs of a blank firmware, with only WiFi configured and default pin states for  pins 0,9,14 configured as low, the rest is default.

### Steps to reproduce
* Configure GPIO 0, 9 and 14 as default low
* Reboot
* System will continue rebooting

### Expected behavior
* Configured pins should be low

### Actual behavior
Continuous reboot loop with the following output:
` ets Jan  8 2013,rst cause:4, boot mode:(3,4)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
ŞU

INIT : Booting version: 
FS   : Mount successful
INIT : Free RAM:26632`

### System configuration

Hardware: Nodemcu V2

Software or git version: Latest mega

SloMusti, 2017-06-13:
This issue has been discussed in the forum as well: https://www.letscontrolit.com/forum/viewtopic.php?t=1462

mvdbro, 2017-06-17:
This seems to depend on your ESP module. I have a NodeMCU board that allows me to set both GPIO 9 and 10 to a low of high state without issues. When using DIO, these pins should be free to use...

psy0rz, 2017-06-19:
yeah, I think we should update the wiki and add a list of "usable IO pins" per hardware board. Perhaps here: https://www.letscontrolit.com/wiki/index.php/ESP_Hardware

And maybe we should warn the users that they can not always use all pins, on this page: https://www.letscontrolit.com/wiki/index.php/ESP_Easy_web_interface#Hardware_page where we warn

psy0rz, 2017-06-19:
@grovkillen could you make a start for this?

Grovkillen, 2017-06-19:
@psy0rz I will do that tomorrow. Good idea

Grovkillen, 2017-06-20:
https://www.letscontrolit.com/wiki/index.php/ESP_Easy_web_interface#Hardware_page
and
https://www.letscontrolit.com/wiki/index.php/ESP_Hardware
has now been updated (not alot), is this enough?

SloMusti, 2017-06-20:
That is great @Grovkillen , having a look at the pin table, there does not seem to be any hardware at the moment that supports pin 9 or such. I believe it would be very helpful to add a warning flag directly to the web interface and the question mark with link to wiki. Most people tend to ignore the wiki pages until need to debug.

Grovkillen, 2017-06-20:
Yes I know (I used to be that guy myself before I got involved in the project). :) I have had problems with GPIO 9 as well and tend to not use that pin at all even though I only experienced the problem with a NodeMCU unit and not WeMos. I think the WebGUI should also state this but I don't know the technical behind the scenes on that one (if it is not doable for some reason). But wiki-link is a need-to-have.

SloMusti, 2017-06-20:
The code that display pin config is here: https://github.com/letscontrolit/ESPEasy/blob/8d2303db5f0ef4cc822524fb2f594d57b38ba83f/src/WebServer.ino#L1110

with this the help button can be added:

`addHelpButton(reply, F("SleepMode"));`

psy0rz, 2017-06-20:
@Grovkillen  thanks! I see there also is https://www.letscontrolit.com/wiki/index.php/Configuration . Maybe we should create some links between that page and https://www.letscontrolit.com/wiki/index.php/ESP_Hardware ?

regarding the milestones map: its nice, but i would prefer to have that info on one place, preferably github. altough github doesnt seem to allow nice formatting, and doesnt allow everyone to update the list. so i'll just edit the wiki with some links to relevant github stuff. :)

Grovkillen, 2017-06-20:
@psy0rz I made the connection from Configuration --> ESP Hardware. 

I understand, and agree, that the official GitHub way is the single source of information (since that is where the action happens) but many users will never read on GitHub thus my attempt to have a list on the wiki as well. Your added link to that page is good though. I will try to do the updates if that is okay, and the devs (you included) can focus on GitHub milestones.

psy0rz, 2017-06-20:
offcourse thats ok :) thanks!

Grovkillen, 2017-06-22:
Should we close this issue?

closed 2017-06-23 by psy0rz