letscontrolit/ESPEasy #358: now() method returns non-utc epoch time, when there is non-zero ntp offset.

rozpuszczalny, 2017-06-14:
### Steps to reproduce

1. Configure NTP Time
2. Set non-zero NTP Offset
3. Use `now()` function somewhere

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
It occurs everytime.

### Expected behavior

`now()` function should return UTC epoch time

### Actual behavior

`now()` function returns UTC epoch time plus ntp offset.

### System configuration

Hardware: ESP12F

Software or git version: HEAD of mega branch (`5136d47`)

### Proposal

I propose to create `now_utc()` function. This function can call `now()` function, but then substract ntp offset from result value of `now()` function. I can make pull request for this - it's really easy to create. When one needs to get time with offset, then one will call `now()` function, but when one needs to get time without offset (so a UTC Epoch Time), then will call `now_utc()` function.

mvdbro, 2017-06-17:
Who has defined that our now() function should return UTC instead of local time?
Anyway, all time stuff within ESP easy assumes local time, never heard of any issue on that...

psy0rz, 2017-06-19:
indeed...this seems like a non issue.

closed 2017-06-19 by psy0rz