letscontrolit/ESPEasy #361: How to send login password for ESP Easy module using mobile app?

foadem, 2017-06-15:
I'd like to create an android app to control my ESP Easy modules. 
Since I have defined login password for my modules I want to login to the server of the ESP Easy without the need of opening any webpage or webview within my app.
Is there an http command to access the ESP Easy server which contains the login password?

rozpuszczalny, 2017-06-15:
You can try to make HTTP POST request to /login with payload `password=[your password here]` and header `Content-Type` set to `application/x-www-form-urlencoded`. On my testing gear it went like this:
`POST /login HTTP/1.1
Host: 192.168.1.106
Content-Type: application/x-www-form-urlencoded

password=123qwe`
After sending this request, ESP respond with 200 OK and I was able to manipulate ESP.

Keep in mind that there's no such a thing like session management on ESPEasy, so once you unlock panel, anyone can access to this panel.

foadem, 2017-06-15:
Thank you so much for your answer.
If I want to send this request in RESTClient addon in FireFox, How can I do this?
Sorry if I'm beginner in networking

rozpuszczalny, 2017-06-15:
I'm not quite sure, because I mainly use Chrome with Restlet Client, but you should see somewhere a select with HTTP request methods (eg. GET, POST). Choose POST method. Then, in the address input, fill `http://[ip of esp]/login`. Next, in request headers, add `Content-Type` header as mentioned above, and finally, in request body area type `password=[your password here]` (without square brackets, of course). Then find somewhere a send button and you're done. Hope it helps, if not, find online a tutorial about HTTP requests or tutorial on usage of REST Clients.

foadem, 2017-06-17:
Thank you so much.
I could send request.

mvdbro, 2017-06-17:
The issues section is meant for reporting bugs, please ask support questions in our community forum
https://www.letscontrolit.com/forum/index.php

closed 2017-06-17 by mvdbro