letscontrolit/ESPEasy #362: _P038 plugin is broken

papperone, 2017-06-15:
### Steps to reproduce
How can we trigger this problem?
In device if you select "NeoPixel Basic" (_P038) the GPIO value is not saved thus plugin is unusable

Does the problem presist after powering off and on? (just resetting isnt enough sometimes) YES

### Expected behavior
Tell us what should happen? I was expecting the plugin to be woking as part of the basic packages

### Actual behavior
Tell us what happens instead? As per above Plugin is unusable/broken

### System configuration

Hardware: tested on plain ESP8266-12 and WEMOS-D1-MINI

Software or git version: 2.0.0-dev10

papperone, 2017-06-15:
Fixed by myself.
As I'm quite new to GITHUB and I've no clue how to submit the plugin I've fixed, can someone help me?

jkDesignDE, 2017-06-15:
see [wiki](https://www.letscontrolit.com/wiki/index.php/Tutorial_preparing_to_work_with_github)

papperone, 2017-06-16:
I think i've done it! How can I check if I did any mistake? Sorry to bother but even thou I'm a coder I've never used GitHub nor other code sharing tools :(

closed 2017-06-16 by papperone