letscontrolit/ESPEasy #367: Wrong values from DHT11 when using long time intervals in sleep mode

tituszx, 2017-06-17:
### Steps to reproduce
Using Sleep mode with 600 s period and DHT11 and DS19B20.
### Expected behavior
Constant or little differ values in same conditions from DHT11
### Actual behavior
Wrong values from DHT11 when using long time intervals in sleep mode.
Sometimes values from DHT11 differ from previous and from DS18B20 on 2-4 grad.
Often, next readings values become normal. 
DS19B20 values are normal.
### System configuration
Hardware:
NodeMCU ver 0.1
Software or git version:
git version 2.0.0
### Temporary solve
Waste first reading from DHT11, and use the second reading.
In DHT module
` pinMode(Plugin_005_DHT_Pin, OUTPUT);  // DHT start condition, pull-down i/o pin for 18ms 
  digitalWrite(Plugin_005_DHT_Pin, LOW);              // Pull low
  delay(18);
  digitalWrite(Plugin_005_DHT_Pin, HIGH);             // Pull high
  delayMicroseconds(20); // was 40
  pinMode(Plugin_005_DHT_Pin, INPUT);                 // change pin to input
  delayMicroseconds(10);`
change to:
` pinMode(Plugin_005_DHT_Pin, OUTPUT);  // DHT start condition, pull-down i/o pin for 18ms
  digitalWrite(Plugin_005_DHT_Pin, LOW);              // Pull low
 delay(18);
 digitalWrite(Plugin_005_DHT_Pin, HIGH);             // Pull high
 delayMicroseconds(20); // was 40
 pinMode(Plugin_005_DHT_Pin, INPUT);                 // change pin to input
 delay(1200); // waste first sample
 // restart DHT
 pinMode(Plugin_005_DHT_Pin, OUTPUT);  // DHT start condition, pull-down i/o pin for 18ms 
 digitalWrite(Plugin_005_DHT_Pin, LOW);              // Pull low
 delay(18);
 digitalWrite(Plugin_005_DHT_Pin, HIGH);             // Pull high
 delayMicroseconds(20); // was 40
 pinMode(Plugin_005_DHT_Pin, INPUT);                 // change pin to input
 delayMicroseconds(10);`

uzi18, 2019-09-04:
@TD-er maybe we need something like "PLUGIN_WAKE" or option to globally ommit 1st READ

TD-er, 2019-09-04:
If it is plugin specific, the plugin should decide if the sample is valid and only return success=true in the PLUGIN_READ function if it is valid.

uzi18, 2019-09-04:
how to know we are just after wake?

TD-er, 2019-09-04:
Well as a plugin you know when the sensor is initialized.
So you can count the number of samples taken.
That's why I added some generic approach for having local memory for a plugin instance. (not using static variables, which are shared among all instances of the plugin)
Look for `PluginTaskData_base` in the code.

closed 2019-10-11 by Grovkillen