letscontrolit/ESPEasy #369: Support for "TTP229" for multiple switch inputs (up to 16 keys)

Grovkillen, 2017-06-19:
As proposed in [this thread](https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3224&p=16957#p16957) it would be nice to have support for the TTP229 circuit in order to have a large number of switches for one ESP. The board is cheap and uses i2c for multiple switches.

[General list of TTP229 boards found on AliExpress](https://www.aliexpress.com/wholesale?catId=400103&initiative_id=AS_20170619005237&SearchText=ttp229)
[A specific board that I think looks nice to use](https://www.aliexpress.com/item/16-Keys-Capacitive-touch-TTP229-module-I2C-bus/32672496161.html?spm=2114.01010208.3.33.TFhINn&ws_ab_test=searchweb0_0,searchweb201602_5_10152_10065_10151_10208_10068_10301_10136_10137_10303_10060_10155_10062_437_10154_5300012_10056_10055_10054_10059_303_100031_10099_5320013_10103_10102_10096_10052_10053_10142_10107_10050_10051_10171_10084_10083_10080_10082_10081_10110_519_10111_10112_10113_10114_10182_10078_10079_10073_10123_10189_142-10051_10112,searchweb201603_16,ppcSwitch_5&btsid=4e3e0bdb-ce96-476e-8d00-fa6bd76aff3d&algo_expid=44c79530-e448-45da-9c6c-b84579492e97-4&algo_pvid=44c79530-e448-45da-9c6c-b84579492e97)

Not necessary to have in 2.0.0, just a nice to have. :+1:

jkDesignDE, 2017-06-19:
Little correction:
 - The variant used on china boards have NO! I2C
 - It is not for switches - only touch pads

You can test it - see Pull requests

Grovkillen, 2017-06-19:
Ah, I understand. Will have to wait a few weeks for mine to arrive. Thanks for the plugin!

Grovkillen, 2017-06-19:
@jkDesignDE and @psy0rz I close this one. JK had set a new world record of request-to-commit-to-merge! Awesome in every aspect of the word :)

Grovkillen, 2017-06-19:
#370 and closed

closed 2017-06-19 by Grovkillen

psy0rz, 2017-06-19:
hahaha ok :D