letscontrolit/ESPEasy #371: Notifications wish list

Grovkillen, 2017-06-19:
I use Pushover for notifications. Any chance to get their API implemented for notifications?

Other solutions with the same functions (I don't mind switching if one API is better then the next):

https://pushover.net
https://www.pushbullet.com/
https://pusher.com/push-notifications

[Thread on forum](https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3239)

Just a future wish for enhancement.

mvdbro, 2017-06-19:
Unfortunately, we have to forget anything that uses SSL with ESPEasy on ESP8266. It uses a huge amount of RAM and will crash instantly...
You can use small sketches with SSL but ESP Easy isn't small enough...
This would need ESP32 i guess...

psy0rz, 2017-06-19:
yeah maybe we can look into it in the future when there is better ssl support in wificlient or someother perhaps. (it also seems that storing all the root certificates in memory is a problem, so someone would have to create a way to do it via spiffs or something?)

Grovkillen, 2017-06-19:
Aha, thanks for clearing that.

psy0rz, 2017-06-19:
sorry that there's no easy fix for it now :(

Grovkillen, 2017-06-19:
No problem, I just thought it would be nice to have. I'll use it on controller side for now. Just thought it'd be a good way of reporting any issues with server/controller/broker from the node side (since pushover also have a possible feedback thing). But let's focus on more to the core fixes as for now. Good job everyone!

psy0rz, 2017-06-19:
yeah it would be a perfect feature for a notification plugin :)

Grovkillen, 2018-11-10:
This is open for future ESP32 implementation...