letscontrolit/ESPEasy #377: for the HLW8012 plugin I need to store double values as plugin parameters

papperone, 2017-06-21:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
To calibrate the HLW8012 chip I need to keep as settings 3 values in DOUBLE but converting them to FLOAT I loose decimal digits

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
YES

### Expected behavior
Tell us what should happen?
Expeting to be able to keep all decimal digits of a DOUBLE parameter 

### Actual behavior
Tell us what happens instead?
As values of plugin settings are stored as FLOAT, the more integer digits they do have the less decimals:
- value     1234.456 is stored properly
- value 123456.789 is stored as 123456.8 (rounded to 1st decimal)
- value 12345678.123 i stored as 12345678.00 (rounded to first integer)
for at least 2 parameters it's important to have DOUBLE precision as they are greater than 1.000.000 and with at least 2 decimal digits

### System configuration

Hardware: does not matter, it's a software issue

Software or git version: latest GIT

jkDesignDE, 2017-06-21:
You can do some C-magic like:
```
        double* da = (double*)Settings.TaskDevicePluginConfigFloat[event->TaskIndex];
        da[0] = 1;
        da[1] = 2;
        double* db = (double*)Settings.TaskDevicePluginConfigLong[event->TaskIndex];
        db[0] = 3;
        db[1] = 4;
```
You can use 4 double values BUT don't use TaskDevicePluginConfigFloat and TaskDevicePluginConfigLong for other floats/longs.

Implementing 3additional doubles costs 288 bytes RAM

papperone, 2017-06-21:
Sorry but I'm not familiar with that type of hack above... Can you exaplin me better how they do work and how I shoudl be using them correctly to store the 3 DOUBLE values?

jkDesignDE, 2017-06-21:
See the float-array and the long-array as piece of memory with 16 bytes each. For a double you need 8 bytes. --> two doubles in each piece of memory.
The magic is only to cast from a pointer to float/long to a pointer to double.
Usage:
If you want to access to your variables (read or write) start with the both "double* d..." lines in your scope. Then you can write with `dF[0] = 1.234567;` or read/calc with `a = b * dF[0];` (for 1st var)

papperone, 2017-06-21:
ok got it :)

jkDesignDE, 2017-06-21:
You can also use
`void LoadCustomTaskSettings(int TaskIndex, byte* memAddress, int datasize)`
`boolean SaveCustomTaskSettings(int TaskIndex, byte* memAddress, int datasize)`
Usage see P023_OLED.ino

psy0rz, 2017-06-22:
that later suggestion is a bit cleaner and the preferred way, since its meant for custom stuff.

closed 2017-06-22 by psy0rz

papperone, 2017-06-22:
Yes indeed, I'm testing the CustomSettings functions as it seems the best option! thanks jkDesignDE