letscontrolit/ESPEasy #380: Just a clarification for wiki

Grovkillen, 2017-06-22:
Is the unit number spectrum (1-31) used for anything more than UDP communication? According to the [page for config](https://www.letscontrolit.com/wiki/index.php/ESP_Easy_web_interface#Config_page) on wiki it is stated that: 

`This is important for MQTT connections and the Node list!`

But I don't understand why this should have anything to do with MQTT? Don't we mean UDP? And if we're not to use UDP do we still need to obey the 1-31 rule for unit number?

Just need to know since I myself do not understand the limitation :)

psy0rz, 2017-06-22:
for MQTT it has to do with the client id, every node that connects to the broker has to send an unique clientid: https://github.com/letscontrolit/ESPEasy/blob/mega/src/Controller.ino#L104

its also used for Inter-ESPEasy Network: globalsync and the unit list.

Grovkillen, 2017-06-22:
Aha, I understand. But this should not be restricted to only use 1-31? (If using only MQTT as communication?)

"Inter-ESP Easy Network: globalsync and unit list" is that used generically or is it only optional?

Grovkillen, 2017-06-22:
HiveMQ [says this](http://www.hivemq.com/blog/mqtt-essentials-part-3-client-broker-connection-establishment) regarding this topic: 

`**ClientId**
The client identifier (short ClientId) is an identifier of each MQTT client connecting to a MQTT broker. As the word identifier already suggests, it should be unique per broker. The broker uses it for identifying the client and the current state of the client. **If you don’t need a state to be hold by the broker, in MQTT 3.1.1 (current standard) it is also possible to send an empty ClientId**, which results in a connection without any state. A condition is that clean session is true, otherwise the connection will be rejected.`

So it could be empty...

papperone, 2017-06-23:
That will probably hit me in a while as at home I've already around 20 nodes (plus 3 or 4 which are not runnign ESPEasy due to the specific tasks they are doing that was impossible to implement them in ESPeasy framework) and with future expansion of my home automation system I'll be needing more than 31 units all running ESPEasy :( any possibility to open a request for 2.1.0 milestone to go to at least 255 devices using still a byte as identificator?

Grovkillen, 2017-06-23:
Me and some friends has experimented with unit numbers above 31+ and no issues has arise as far as we can see. We only use MQTT to communicate. But I'd like to wiki-write the official limits. Don't want end users to be left in the unknown.

papperone, 2017-06-23:
Thanks, good to know!!! As I do use only MQTT and all the rest is done in NodeRed.

Grovkillen, 2017-08-24:
I have now updated the wiki-text and vote for this to be closed: https://www.letscontrolit.com/wiki/index.php/ESP_Easy_web_interface#Config_page

I refer to this text found in source: https://github.com/letscontrolit/ESPEasy/blob/mega/src/ESPEasy.ino#L227

> // Only relevant for UDP unicast message 'sweeps' and the nodelist.

closed 2017-08-24 by Grovkillen