letscontrolit/ESPEasy #381: MQTT Broker not reconnecting using dynamic url (if IP is changed)

Grovkillen, 2017-06-22:
Me and my friend has both experienced trouble when the broker IP address has changed behind a dynamic dns url address.

For example homeaddress.dyndns.org redirects to 10.0.0.1 and after a power down the broker get a new ip address 10.0.0.254, this is reported to dyndns but the unit still try to connect to the old 10.0.0.1 address.

Is my suspicion correct; that the ESP Easy code takes the dynamic dns address, after connecting only use the direct IP address and never recheck the dyndns address and see if a new IP is needed.

This should be done if the broker is not responding.

tedenda, 2017-06-22:
Is this "related" to this #261 issue?

Grovkillen, 2017-06-22:
Yes it probably is related.... seems like it.

pwrsoft, 2017-10-07:
The same issue, nearly 6 month passed - could someone knowledgable fix it, please?!

Grovkillen, 2017-10-07:
I've added it to 2.1.0 since the 2.0.0 is pretty much closed and only smaller issues are being fixed.

TD-er, 2017-10-07:
And also related to #521 

I was just thinking, maybe if a TCP request is being performed and the other side is not responding, perform a DNS update every now and then (at most once per 10 minutes or so).
That may be a workable workaround?

pwrsoft, 2017-10-09:
@Grovkillen thanks for adding it to the 2.1.0 milestone list.
Actually I have 2 nodes with different firmware, where this bug is present: it's master and mega versions.
It would be great if it will get fixed in both branches.

Grovkillen, 2017-10-09:
@pwrsoft the mega will become the master when the FW is released as 2.0.0

ziolek, 2017-10-28:
I'm experiencing related issue every time [CloudMQTT](https://www.cloudmqtt.com/) changes IP for it's service behind the same hostname.
It looks like the the name is being resolved only once after setting hostname in MQTT config.

Version: **v2.0.0-dev12**

travellingnashes, 2017-10-30:
Can we get a patch for this against 2.0.0 dev 12 please. Ive had 9 sensor boards go offline due to this only being looked up once when the settings are stored. I would like to test the patch and ensure it closes the issue we are discussing. If not I need to move from ESPEasy to custom code asap

TD-er, 2017-10-30:
I've read all related issues and it seems it is only fixed for C11, the http-controller: #302
But it should be fixed for every network-connection, thus including all controllers, syslog server, etc.
And perhaps there is also some plugin which uses network connections? (not sure)

I will look into it this evening to see how wide spread this issue is.
Best would be to have some cache in memory for every reference to a hostname and thus prevent unneeded writes to flash when an IP changes. This cache would be only 4 bytes in size. Only problem is to combine them with the hostname. So perhaps if we can have some 1 byte (unique) identifier along with every hostname, we can have a lookup table present in memory.

When the impact is clear, a well informed decision can be made.

Walderbash, 2017-10-30:
I was just looking into this last night trying to connect to Bluemix IOT. It looks like ESPEasy does a DNS lookup when you save the MQTT config, and when doing an actual mqtt call always uses the saved IP address instead of the hostname.

[image]
change to something like this:
[image]

Wouldnt that solve the issue? (dont know if there is an other DNS cache)

travellingnashes, 2017-10-30:
No thats not forcing a reconnect when public IP changes. I will add some diagnosis and try to understand what it is doing. Logic looks sound though

Mike

travellingnashes, 2017-10-30:
Sorry my error. Yes that is working against the test that I am doing. Not sure against the efficiency of this but it is fine for my need as I am only doing this once an hour per device. Great thanks

Mike

TD-er, 2017-12-06:
I just looked into the code of the Arduino library.
The code for DNS lookups ( https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip/src/core/dns.c ) is using a table to cache the requests.
So apart from having to lookup the cached result, there shouldn't be any reason to keep a local cached IP-address in every controller instance.
Only thing is that the DNS_TABLE_SIZE might be too small for global use.

My suggestion is to add a helper class to be used in ESPeasy, which will cache only the number of hosts requested in that instance of the helper and only so long as (global) defined or until a timeout occurs.
Then use this helper class anywhere a WiFiclient is used in the code and thus fix all controllers at once.
This way we have a single instance of code to tweak and tune and all plugins and controllers will profit from those fixes.

psy0rz, 2017-12-08:
cant we just do a dns lookup every time, since it already does caching?

psy0rz, 2017-12-08:
Yeah I like this way a lot, I think we should keep it that way. 

So with this PR "Use DNS" actually means: "Resolve hostname everytime" vs "Resolve hostname only on save"?

closed 2017-12-08 by psy0rz

TD-er, 2017-12-08:
Yep, that's actually what's changed :)
And also, it now works like this for all controllers and the Email module and the MQTT import plugin.

There are still things to improve, but that's for later. (like why store IP and hostname)
That will result in change how things are stored in flash, so that's definitely not for 2.0

psy0rz, 2017-12-08:
Indeed, thanks! @Grovkillen is there a wiki of the settings page? Maybe we could explain what "Use DNS" does from now on:

"Use DNS": Lookup the IP address every time ESPEasy connects to the host.

Disable this function to only lookup the IP address once when you save the settings.

TD-er, 2017-12-08:
@psy0rz When you set the toggle to "Use IP", the IP lookup of the entered (and saved) DNS hostname is displayed and only the IPaddress will be used.
When you switch back to "Use DNS", sadly the original hostname is not being displayed anymore.
That's something to look into for later releases and perhaps redesign of that feature. (why store hostname and IP)

psy0rz, 2017-12-08:
ah right i understand.

Grovkillen, 2017-12-08:
I will update Wiki this weekend.

catcam, 2017-12-09:
Can you please post link to compiled  v2.0.0-dev13 ... im not lazy but im on field and this will save me lot of time ... and tnx for great job

Grovkillen, 2017-12-09:
@catcam No dev13 is released yet. You find the released packages here:

https://github.com/letscontrolit/ESPEasy/releases

catcam, 2017-12-09:
yes ... but looking for version with dns resolve fixed  ... again tnx for quick reply

Grovkillen, 2017-12-09:
@catcam as i said, the dev13 is not out yet. Psy0rz just mentioned it for my wiki text that I will be adding later today. But again, the dev13 is not yet released.

catcam, 2017-12-09:
ok it is clear for me now ... thumbs up and looking forward to dev13