letscontrolit/ESPEasy #386: IRTX every fitht bit is zero on raw send

1500WK1500, 2017-06-24:
### Steps to reproduce
If I want to send for example a long pulse with  16 bits (FFFF). 
http://192.168.1.140/control?cmd=IRSEND,RAW,ffff,38,400,400

### Expected behavior
I would expect that output for FFFF is:
"Interpreted RAW Code: 1111111111111111"

### Actual behavior
However after every fourth bit a zero is added. (expect for the last bit)
"Interpreted RAW Code: 1111011110111101111"

### System configuration
Software or git version:
ESPEasy_v2.0.0-dev10_normal_4096.bin

### Solution
I could allready fix this issue if I change the line https://github.com/letscontrolit/ESPEasy/blob/0404a87/src/_P035_IRTX.ino#L131
from
`if (c0+c1 != 0) {` 
to
`if (shft != 1) {`

Should I create a pull request for this issue?

psy0rz, 2017-06-25:
its not really clear to me those 2 loops are trying to accomplish. (zero comments and cryptic variables names) pretty bad code.

so i cant decide if you're actually fixing it or if you might be creating a new problem as well.

psy0rz, 2017-06-25:
also there is no explanation of what kind of input this plugin expects at https://www.letscontrolit.com/wiki/index.php/IRTX

psy0rz, 2017-06-25:
@GusPS could you add some commands and/or examples to the code, so that other people can understand whats going on? :)  Now it would take a lot of time to reverse engineer whats going on.

GusPS, 2017-06-28:
No problem @psy0rz. I will try to help with this bug as soon as I can and add some comments.

psy0rz, 2017-06-28:
@GusPS thanks! :+1:

GusPS, 2017-06-28:
Hi @1500WK1500,

Remember that "FFFF" is a Base32 string, so translating it to binary, it will be 01111011110111101111. The left-most 0 is supressed because is not used (is just like sending no ir-light).

So, the "Actual behavior" is really the Expected.

As I wrote on the forum, the idea is to encode it as a base32 to reduce the URL length. This b32 string is then converted into 0s and 1s, which each one of them will be sent with the specified ms by the TX plugin.

char c = ((IrRaw[i] | ('A' ^ 'a')) - '0') % 39;   <--- This is the character value from 0 to 31 [see https://en.wikipedia.org/wiki/Base32#base32hex]

c1 and c2 are the vars that counts the 0s and 1s. The idea is to count how many consecutives 0 and 1 are, and finally multiply it by the length of the pulse or space and add it to the buffer that will be used by the Plugin_035_irSender->sendRaw(buf, idx+1, IrHz);.

I will be adding comments on code, but feel free to change it to be more accurate. I speak spanish so sometimes the way I explain it could be improved.

And, let me know if you have any other doubt or suggestion.
Regards!

1500WK1500, 2017-06-29:
@GusPS 
Thanks for the explanation of the code! I totally overlooked that the string should not be in "normal" hex representation.

As mentioned from @psy0rz an explanation of the command syntax would also be very helpful on the wiki.
[https://www.letscontrolit.com/wiki/index.php/IRTX](url)

closed 2017-06-29 by 1500WK1500

psy0rz, 2017-06-29:
@GusPS  thanks for taking time to elaborate and cleanup the code. :)

ShardanX, 2017-07-07:
About the command for IRTX i know only the old (R14x) version.
Syntax:
control?cmd=IRSEND,<Protocol>,<Hexcode>,<Length>
Protocol: NEC, JVC, RC5, RC6, SAMSUNG, SONY, PANASONIC
Hexcode: The code to send
Lenght: the bit length of the code.

Soe info is in the thread [(https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=328&hilit=IRSEND)]

I've used an Arduino with the "Arduino-IRremote 2.1.0" and a TSOP-Receiver to decode my remotes, works quite nice.