letscontrolit/ESPEasy #397: RTTTL not compatible?

kuipers250, 2017-07-01:
I am using 2.00-10

The problems seems to be that not every RTTTL will play correctly. I found a lot of sounds on  http://www.picaxe.com/RTTTL-Ringtones-for-Tune-Command/

And a lot/most won't play correct on the ESPeasy. Like this one.

d=4,o=5,b=200:g#,c#,8p,c#6,8a#,g#,c#,8p,g#,8f#,8f,8f,8f#,8g#,c#,d#,2f,2p,g#,c#,8p,c#6,8a#,g#,c#,8p,g#,8f#,8f,8f,8f#,8g#,c#,d#,2c#

Or on http://www.vex.net/~lawrence/ringtones.html for example the Mission Impossible theme

d=32,o=5,b=125:d6,d#6,d6,d#6,d6,d#6,d6,d#6,d6,d#6,d6,d#6,d6,d#6,d6,d#6,d6,d#6,d6,d#6,e6,f#6,d#6,8g.6,8g.,8g.,8a#,8c6,8g.,8g.,8f,8f#,8g.,8g.,8a#,8c6,8g.,8g.,8f,8f#,16a#6,16g6,2d6,16a#6,16g6,2c#6,16a#6,16g6,2c6,16a#,8c.6,4p,16a#,16g,2f#6,16a#,16g,2f6,16a#,16g,2e6,16d#6,8d.6

psy0rz, 2017-07-01:
the rtttl code is not really great and probably contains some bugs.

kuipers250, 2017-07-02:
Like the functionality for my purpose. So when there can be enhancements I would certainly appreciate it.

grabula, 2018-02-19:
I found that song stops playing on # symbol, if you remove them it is ok, but they do not sound as expected. Songs that do not contain # play ok

uzi18, 2018-03-01:
try to replace # with - and then send to play

uzi18, 2018-03-16:
I will try to find some hardware to test if it realy works and how ...

fmuntean, 2020-05-22:
I just checked the code for it and I don't see anything wrong.
However the code expects a song name always.
while(*p != ':') p++;    // ignore name
Due to this anything up to first colon is ignored.
So the correct way of playing these two songs is to prepend the following:
rtttl,<pin>,<name>:

If there is no name then the code will skip the defaults as d=4,o=5,b=200: and will use the internal defaults thus the songs will not be played correctly.

psy0rz, 2020-05-28:
Thats a prime example of a serious bug:
```
while(*p != ':') p++;
```

This means: if there is no : in the string the code will loop forever and eventuel crash with a memory exception error.

There are many such instances in that code if i remember correctly. Change your code to be fool proof, so it can never crash and returns a nice error if something is wrong.

Do this with strnchr for example, instead of writing your own loops.

fmuntean, 2020-07-28:
> Thats a prime example of a serious bug:
> 
> ```
> while(*p != ':') p++;
> ```
> 
> This means: if there is no : in the string the code will loop forever and eventually crash with a memory exception error.
> 
> There are many such instances in that code if i remember correctly. Change your code to be fool proof, so it can never crash and returns a nice error if something is wrong.
> 
> Do this with strnchr for example, instead of writing your own loops.

Feel free to create a pull request and improve the code please.

psy0rz, 2020-07-28:
> Feel free to create a pull request and improve the code please.

Lol..at that time i WAS the maintainer of this project. :)

tonhuisman, 2023-09-02:
The 'serious bug' in the code seems to be fixed, so this can be closed.

closed 2023-09-02 by TD-er