letscontrolit/ESPEasy #398: BMP085/180 wrong sensor readings

igrcic, 2017-07-01:
Hello,

the bmp180 sensor  seem to have problems with temp readings. It shows temperature of 117C when it should actually be around 25C. The value seems to be ok on reboot  but then it jumps to wrong value after a few readings.

### Steps to reproduce
How can we trigger this problem?
Connect the sensor and wait for a few readings.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes, even after powering off.

### Expected behavior
Show the correct temperature.

### Actual behavior
Shows the wrong temp :)

### System configuration

Hardware:
Wemos D1 mini (Esp-12S)

Software or git version:
v2.0.0-dev10

igrcic, 2017-07-01:
This is probably related to:
https://github.com/nodemcu/nodemcu-firmware/issues/1471

ShardanX, 2017-07-03:
I've made a test curcuit using BMP180, breadboard and nodeMCU with ESP-12E.
Selfcompiled ESPEasy Mega with sources of 26th of June, 2017 (normal).
It's up and running for 24 hours now showing correct temperature and pressure.
I'm not able to reproduce the issue.

closed 2018-07-24 by TD-er