letscontrolit/ESPEasy #399: Use device name as AP SSID

SloMusti, 2017-07-01:
Using the standard `ESP_<unit number>` SSID is impractical for applications when we wish to have a more user-friendly setup with convenient naming. I suggest device name is used in this case or at least it can be user configurable.

Relevant code: https://github.com/letscontrolit/ESPEasy/blob/ea5168c08a745c56ec10c237ef4cfac86e4cc227/src/Wifi.ino#L9

psy0rz, 2017-07-01:
well maybe, in some future version. i'm not sure though how people feel about this and if there is some kind of reasoning behind the current system.

@mvdbro  what do you think?

Grovkillen, 2017-07-03:
Or at least we might call it "ESP_Easy_AP" or something as default? We need to push as much as possible that the FW is ESP Easy and not a generic FW. Strengthen our brand. :)

Grovkillen, 2017-07-26:
This should be closed.

closed 2017-07-29 by psy0rz