letscontrolit/ESPEasy #400: Enhance documentation on Rules

SloMusti, 2017-07-01:
To enable clearer use of Rules beyond what is currently documented here https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules a more detailed documentation would be helpful.

A few examples worth clarifying:

 * if `timerSet` is called for an active timer prior to its expiry, then the timer counter is reset
 * doing composite logical operations, such as AND/OR between multiple values (better explanations on creating custom events to avoid nesting)
 * details on deepSleep though rules

I invite others to post things requiring clarification here.

Grovkillen, 2017-07-03:
Yes I'm onto this. Will make a "better" rules page. Maybe we could use this topic for adding more stuff that could be explained on that refurbished page?

Just add a comment here and I'll try to get this done and online within the month of July.

psy0rz, 2017-07-03:
w00p thanks @Grovkillen

Grovkillen, 2017-08-28:
I have more started this with:

https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules#Syntax

I will add more to it...

beicnet, 2017-08-28:
@Grovkillen, Nice, add a bunch of random example to it! ;)

psy0rz, 2017-09-14:
@Grovkillen  thanks looks awesome now!

closed 2017-09-14 by psy0rz

iBobik, 2018-05-05:
I miss information how big the timer argument could be. 99999999999 did not worked (timer was not changed).