letscontrolit/ESPEasy #401: deepSleep from rules, some task-values are not saved

SloMusti, 2017-07-01:
### Steps to reproduce
 * Run a firmware post this pull request: https://github.com/letscontrolit/ESPEasy/pull/375
 * Enable Rules
 * An example of a rule:
```
On Rules#Timer=1 do  // going to sleep
 TaskValueSet,1,1,1
 deepSleep,600 // go to sleep for 600s
endon
```

### Expected behavior
TaskValueSet should set the values such that it is persistent during deepSleep, like other values that are written a while prior to deepSleep

### Actual behavior
TaskValue defined in such a way is not saved prior to deepSleep and is not restored. A fix is required to make sure all things are diligently completed before going to sleep.

### System configuration

Hardware: Nodemcu

Software or git version: post d09c97073eef6849d22aac83c6c050a2dc65747b

SloMusti, 2017-07-01:
@psy0rz do you maybe have an idea how the logic for saving task values works, such that I can fix this bug.

psy0rz, 2017-07-01:
i think the saving of that stuff happens every 'x seconds' or something like that. we should probably change that so it just happens once in deepsleep.

SloMusti, 2017-07-02:
@psy0rz I think it is ok that it happens every X seconds, but also need to explicitly call it in deepSleep routine prior to going to sleep.

psy0rz, 2017-07-02:
indeed

SloMusti, 2017-07-03:
Interestingly, this is implemented already: https://github.com/letscontrolit/ESPEasy/blob/mega/src/Misc.ino#L65

I need to further verify this happens properly if called.

psy0rz, 2017-07-03:
hmm strange..might be a bug

SloMusti, 2017-07-28:
So far have not observed it again, but using plenty of delays, needs proper testing with sleep right after value is changed.

Grovkillen, 2018-10-23:
Is this still an issue?

TD-er, 2018-10-23:
I guess so