letscontrolit/ESPEasy #408: HX711, load cell (plug-in needed?)

Grovkillen, 2017-07-03:
I just found this one month old forum topic.

Do we need a plugin for the HX711 or is ADS1115 enough?

https://www.letscontrolit.com/forum/viewtopic.php?t=1635

beicnet, 2017-07-03:
@Grovkillen Okay, it's a 24-bit Precision A/D Module, but it's specifically meant for Weighing Pressure and its a Dual Channel Sensor if I'm not wrong, personally I have those in the complete set:
[https://www.aliexpress.com/item/10KG-Scale-Load-Cell-Weight-Weighing-Sensor-HX711-Weighing-Sensors-AD-Module/32728749185.html](url)

ShardanX, 2017-07-03:
It will need another plugin.
HX711 has two differential inputs, ADS 1115 has 4 inputs that can be used as four single ended or two differential inputs. HX711 has 24 bit resolution, ADS1115 has 16 bit.
As far as i can see from the Datasheet the interface of the HX711 is serial, not I²C.
So i assume differences are too much to use the ADS1115 plugin.

beicnet, 2017-07-03:
@ShardanX True!

beicnet, 2017-07-03:
@Grovkillen btw...you have a very similar module named ADS1015, it's an 12-bit A/D and it has the same pinout as ADS1115 and it's cost less by around 60-70 cent...

Grovkillen, 2017-07-04:
Thanks for the input, we add this for the future I guess. :)

jkDesignDE, 2017-07-04:
@Grovkillen I read the datasheet and it will be a very simple plugin - you get a plugin in the next weeks...

Grovkillen, 2017-07-04:
Sweet! You're faster then Ali to deliver :)

Grovkillen, 2017-07-08:
This is now implemented, #415

closed 2017-07-08 by Grovkillen