letscontrolit/ESPEasy #413: WiFi reconnect feature missing

SloMusti, 2017-07-05:
Now that one can configure two SSIDs, a reconnect feature would be welcome, such that one can either manually or periodically trigger reconnection to the primary access point. In particular that is critical when we operate in a sleep regime where the ESP wakes up an AP (primary) and while that is offline or such is connected to secondary AP. Then we wish to as soon as possible to get it back.

Currently this is only possible using the following methods:
 * Reboot (which disables GPIOs and as such may disable the external AP device enabled by the ESP module)
 * Rules - creating a reconnect rule and calling it manually or periodically:
```On reconnect do
 WifiDisconnect
 WifiConnect
endon
``` 
and then manually call `<ip>/control?cmd=event,reconnect`

I propose the Reconnect features is implemented under Tools such that when you press connect, reconnection is either forced or a separate feature is added. Manually if you press disconnect, then you can not press connect thereafter because you are disconnected.

psy0rz, 2017-07-07:
Thats a good idea. Also it would be nice if it automaticly tries to reconnect to the primary AP, every X seconds or something. (configurable)

SloMusti, 2017-07-08:
Currently I implement exactly this with Rules:

```
On System#Boot do //on boot
  timerSet,1,3000
endon

On Rules#Timer=1 do 
 Event reconnect
 timerSet,1,3000
endon

On reconnect do
 WifiDisconnect
 WifiConnect
endon
```

Grovkillen, 2017-07-12:
Just a input. The command should be called '''WifiReconnect'''

Grovkillen, 2018-10-23:
We should add this.

Grovkillen, 2018-10-25:
We have discussed this offline and have come to the conclusion that we will have a parameter for the wifidisconnect command.

`wifidisconnect` will work as before
`wifidisconnect,<integer>` will disconnect and turn off the wifi module for <integer> seconds.

TD-er, 2018-10-25:
Only turn off the wifi when its "disconnect" time is below X seconds, or else it will add up to the reconnect time.

Also there are reports on some core versions crashing when disabling wifi. We have to look into that.

riker65, 2019-04-02:
Hi, is this still valid? does
wifidisconnect,10
forces a reconnect independent if ESP went into AP Mode?
[https://espeasy.readthedocs.io/en/latest/Reference/Command.html](url) does not mention the integer any more?

I am confused

Thanks T

TD-er, 2019-04-02:
@riker65 The current implementation also does not use a parameter.
So for now the documentation is in sync with implementation

closed 2023-09-03 by TD-er