letscontrolit/ESPEasy #414: Wiegand / PN532 give different values

ShardanX, 2017-07-05:
**Steps to reproduce**

**How can we trigger this problem?**
Connect a PN532 and a Wiegand34 Reader and use the same tag

**Does the problem presist after powering off and on? (just resetting isnt enough sometimes)**
Yes, even after powering off.

Expected behavior
Show same ID number

Actual behavior
Showing different numbers
PN 532 gives:......3004897467 (10110011000110110001100010111011)
Wiegand gives:..3138919347 (10111011000110000001101110110011)

**System configuration**

**Hardware:**
nodeMCU3.0 (ESP-12E)

**Software or git version:**
v2.0.0 self compiled version (normal) sources from June 30., 2017

I've checked the tag i used with a commercial Mifare Chip Reader.
The value from the PN532 is same the reader shows.
Swapping Wiegand D0 and D1 gives even more deviation from the value.

The readers can be used. It makes parallel use of Wiegand readers and PN532 / other reader systems very difficult as two different codes have to be used and maintained.

psy0rz, 2017-09-21:
the pn532 is 13mhz and the  Wiegand34 is 125khz. how do you manage to use the same tag on both?

psy0rz, 2017-09-21:
do you have a link to the one you have? i'll order one to test it.

ShardanX, 2017-12-10:
The PN532 is a board like this one:
[https://de.aliexpress.com/item/New-PN532-NFC-RFID-V3-module-near-field-communication-support-and-Android-mobile-phone-communication-BOARD/32808511010.html?ws_ab_test=searchweb0_0,searchweb201602_1_10320_10152_10321_10065_10151_10344_10068_10345_10342_10547_10343_51102_10322_10340_10341_10548_10193_10194_10541_10084_10083_10304_10307_10302_10539_10180_10312_10059_10313_10314_10184_10534_100031_10319_10604_10603_10103_10605_10186_10594_10596_10142_10107,searchweb201603_12,ppcSwitch_5&algo_expid=a0ab38fd-495c-4392-8610-8ae887f87f3d-19&algo_pvid=a0ab38fd-495c-4392-8610-8ae887f87f3d&rmStoreLevelAB=0](url)
Wiegand Reader: Wiegand is a protocol, not a frequency. It defines the two-wire diferrential sending of data. Readers are availlable for 125 KHz and 13,56 MHz as well. I was testing a box like this:
[https://de.aliexpress.com/item/Proximity-MF-IC-Card-Reader-Wiegand-26-34-13-56MHZ-Reader-ABS-Shell-Door-Access-Control/32827373480.html?ws_ab_test=searchweb0_0,searchweb201602_1_10320_10152_10321_10065_10151_10344_10068_10345_10342_10547_10343_51102_10322_10340_10341_10548_10193_10194_10541_10084_10083_10304_10307_10302_10539_10180_10312_10059_10313_10314_10184_10534_100031_10319_10604_10603_10103_10605_10186_10594_10596_10142_10107,searchweb201603_12,ppcSwitch_5&algo_expid=cb531530-91ee-4ac6-84ad-bd55431c54d0-41&algo_pvid=cb531530-91ee-4ac6-84ad-bd55431c54d0&rmStoreLevelAB=0](url)

Sorry for late answer, I lost sight a bit as the reader ws planned together with a multidisplay box - as two displays didnt work (and still don't work..) with one ESP I had to stop this project for now.

psy0rz, 2017-12-18:
ok ill order one myself and try to fix it

TD-er, 2018-01-22:
Hmm, looks like the byte order is different:
```
PN 532 gives:......3004897467 (10110011 00011011 00011000 10111011)
Wiegand gives:.....3138919347 (10111011 00011000 00011011 10110011)
```
Per byte is the data reversed.

Naturious, 2018-01-23:
From my point the weigand isn't even giving the same number of bytes:
PN532: 04e5babada5180
Weigand: BBAEE504
And I can see they're starting out reversed in the beginning (04 and E5) but then it changes.

TD-er, 2018-01-23:
You can also read those RFid tags using a mobile phone (with NFC capability) and NXP tools (NXP taginfo).
Perhaps keep that order and see if the output then matches one or more fields in the tag?

@Naturious Are you sure the same parameter is given?
Perhaps one is the TID and the other the EPC Memory?

ShardanX, 2018-01-23:
@TD-er i've tested that with a commercial card reader connected via USB to my PC.
The PN532 reader (red pcb, as shown in the hardware wiki) reads same values as the commercial one,  the wiegand does not.

@Naturious  Sure that you use Wiegand34? Some wiegand readers support only 26 bits, others can be set to 26 or 34 bit code.

TD-er, 2018-01-23:
@ShardanX Can you also test with the NXP tools in Android? After all, NXP (Philips) is one of the biggest companies behind RFid (market leader as they claim themselves)

ShardanX, 2018-01-23:
Nope, sorry. I don't own a NFC phone.

TD-er, 2018-01-23:
OK, then I have to dig into my big box of projects waiting to be started/finished and mount the already ordered RFid reader... 
Will do later this week.

Naturious, 2018-01-23:
I think I'm using Wiegand26..
How do you think I can match the 26-bit and 34-bit codes to each other?

psy0rz, 2018-02-02:
@TD-er if its a byteorder thing, just change it so its the correct value. dont mind backwards compatibility if the id's where actually wrong.

TD-er, 2018-02-02:
@psy0rz But what is the 'correct' order?
And there are apparently different code lengths.

I will look into it, but not sure about the when.
I do have some RFid chips, my phone can read them and I do have such a board laying around... somewhere. Now some time. Hmm maybe AliExpress has some spare time on sale, lets order a bunch of them.

Naturious, 2018-03-05:
I changed into Wiegand 32, now it's giving me this: BAE50488
while PN532 is giving  04E5BABADA5180
for the same Mifare card, any thoughts?
(Notice the first 3 bytes are inverted)

ShardanX, 2019-03-20:
At least the Wiegand-34 is still broken, even with the new patch from #2400 
ATM i have no possibility to test for Wiegand-26