letscontrolit/ESPEasy #417: Visible characters on OLED_64x48

sebastiandinu, 2017-07-08:
### Steps to reproduce
Use a 64x48 pixel i2c oled and set a 20 character string on line 1. 

### Expected behavior
Show 8 characters on line one, show 8 characters on line two and the last 4 characters on line three

### Actual behavior
The display shows the first 8 characters on line one and the last 4 characters on line two. 

### System configuration

Hardware:
Wemos d1 mini v2
Wemos oled shield (64x48)

Software or git version:
Flashed with ESPEasy_v2.0.0-dev10_normal_4096.bin

I think the problem is that the plugin has a default of 16 columns set and does not take into consideration the display size.

Grovkillen, 2018-03-16:
@TD-er is this an issue?

TD-er, 2018-03-16:
Which OLED plugin? The OLED, or OLED_Framed?

TD-er, 2019-11-04:
This does seem to be fixed in #2706 
So I will close this one.

closed 2019-11-04 by TD-er