letscontrolit/ESPEasy #420: systime showing seconds is great but not always wanted.

Grovkillen, 2017-07-09:
Many use the systime to display time on a display. Seconds is not always wanted (three extra characters).

I would like to have the systime plus the possibility to create own strings like:

%syshour%
%sysminute%
%syssecond%

This way we can use %syshour%:%sysminute% to create the old way of systime.

Grovkillen, 2017-07-10:
Possibly the variables could also be used in rules?

If syssecond=0 do
 ... //This would be triggered once per minute
Endon

If sysminute=0 do
 ... //This would be triggered once per hour
Endon

If syshour=0 do
 ... //This would be triggered once per day
Endon

ShardanX, 2017-07-10:
Even more if you use a display with systime on it... usually a display is not updated every second so the seconds look quite funny there.

papperone, 2017-07-10:
this is not a bad idea at all and will make rules more powerful I guess!

Grovkillen, 2017-10-16:
@psy0rz should we close this one now? i updated the wiki page: https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables

Will make it more easy on the eye when i get the time though (the page is pretty "ugly" at the moment).

closed 2017-11-12 by psy0rz