letscontrolit/ESPEasy #421: HLW8012 plugin crashed using "SaveCustomSettings" on POW device

papperone, 2017-07-10:
### Steps to reproduce
How can we trigger this problem? 
Enabling the HLW8012 pluing on an ITEAD SONOFF POW unit

Does the problem presist after powering off and on? (just resetting isnt enough sometimes) 
YES

### Expected behavior
Tell us what should happen? 
calling "SaveCustomSettings" function shoudl save the 3 "DOUBLE" variables needed in the SPIFF file

### Actual behavior
Tell us what happens instead?
calling "SaveCustomSettings" function crashed the module

### System configuration

Hardware: ITEAD SONOFF POW

Software or git version: 2.0.0.dev10 with HLW8012 plugin (TEST)

the same plugin works perfectly on a plain ESP8266 and WemosD1 Mini both with 4Mb memory.
as I don't own a POW I cannot test but in the forum users report plugin crashing ESPEasy on the "SaveCustomSettings" call; the lenght of the Double array is 24 bytes, but the same function called from the LED/OLED plugin with same/more lenght works perfectly; can someone give some hints???

jkDesignDE, 2017-07-10:
`(byte*)&hlwMultipliers   -->   (byte*)hlwMultipliers`

hlwMultipliers is already a pointer

papperone, 2017-07-10:
Sorry I realized after a fresh debug session that issue was not caused by this function :(
My bad, but at least I've found it (in some situation the internal multipliers could be saved before the plug was init thus causing the reboot).

closed 2017-07-10 by papperone