letscontrolit/ESPEasy #423: CSS in latest (self compiled) build not "complying to design rules"

Grovkillen, 2017-07-10:
We need to stay with our fundamental color (#07d), the latest code uses an orange (#d50) color for texts and some parts of the WebGUI. Need to revert them to #07d .

The rest of the WebGUI looks great though, and the speed is lightning fast now 👍

sassod, 2017-07-11:
Why the revert I love  it in orange and it is more readable than tbze blue.
Cant see why would a  color be responsible for slow page loads.

Grovkillen, 2017-07-11:
The speed is related to other code changes, nothing to do with the colors.

Regarding the color. If we change the webgui colors we need to change the wiki color as well. We need to have a coherent front of we are going to make ESP Easy defacto standard. Can't just make changes to the visual design without confirming this with the overall design of the project. I like the color as well but nowhere else the orange color is used and the blue has come to be the branding of ESP Easy. 

If we are to involve more colors a design guide is needed. That might be a part of v3?

psy0rz, 2017-07-11:
yeah i noticed it. who changed it to orange? @jkdesign?

jkDesignDE, 2017-07-16:
Normally a blue color is used for clickable areas (button, link). On menu rework I searched for a harmonic color sheme which includes the given blue and the used grays. With this orange (I think) all fits together and is unique. If majority prefers a pure blue design, it's also OK for me. I think the web-design needs only a colored logo on the first page - no more changes needed.
I see it was changed to blue 2 houres ago - issue can be closed

psy0rz, 2017-07-16:
yep martinus changed it back. i understand your reasoning but lets keep it blue for now.

closed 2017-07-16 by psy0rz