letscontrolit/ESPEasy #426: P050 TCS34725 sluggish at best.

Grovkillen, 2017-07-10:
Any news regarding this plugin? I have the setup but cannot get any good readings and the plugin is grinding the ESP unit to almost a stop.

krikk, 2017-07-10:
more details please, are you sure that this plugin is the cause of slow ESP...i wrote the plugin and tested it with 2 sensors without problems...never had problems with not getting readings...

are you sure you do not have a low RAM issue?

Grovkillen, 2017-07-10:
I can check tomorrow. It could be it (I did mess around looking through the plugin list, that might be the case!). I take my somewhat negative comments back until I have tested it again. 👍

Grovkillen, 2017-07-11:
@krikk I have now tested and the sluggishness came from me flipping through the list on my test unit. The RAM dropped for sure. Sorry for thinking this had anything to do with your plugin.

But I need some more explanation regarding the settings. I cannot get the sensor to both have a close to RGB(0,0,0) for a black fabric and close to RGB(256,256,256) for a white piece of paper. Not without a lot of tinkering.

The best results I got was using 24ms time and 4X gain together with this formula for each RGB **%value%/20000**. We need to add more information on how to use this sensor.

Grovkillen, 2017-07-11:
Since 256 is the highest RGB value we should have an option to click if we do want higher values than this. Default should be stopped at 256 to not confuse the user.

krikk, 2017-07-11:
the max value of RGB is 255, not 256 (only a byte variable) ... led control is implemented already but not merged, see pull request #255 

...the drop down list: yes will implement this... but will take a while... i'm off for a week of bathing and i am not sure if i have time to code there :)

Grovkillen, 2017-07-11:
Ah, yes. 255 it should be. Thanks!

You do it when you are fit for it... much needed off time I suspect! :) Have a nice trip.

Grovkillen, 2017-07-11:
@krikk I just finished the rule. Could you try if possible? Best results is by using dull paper with different colors or fabrics (I went through my kids stuffed animals, they worked out good). High gloss is not to recommend (not since we only use LED analysis and not "one with LED + one without").

https://www.letscontrolit.com/forum/viewtopic.php?f=2&t=3021&p=17512#p17512

PR #3454 (2021-02-01): [P050] Additional output settings
- Added ~~minimal~~ updates to Adafruit library (including updated comments)
- Added Color Temperature DN40 calculated value
- Added Lux calculated value
- Added Clear output value
- Added Normalized RGB values (0..255)
- Added RGB Transformation matrix (3x3)
- Added optional events for calculated and measured values
- Cleaned up the existing options

TODO:
- [x] ~~Documentation~~ (done separately)
- [x] Adjust/fine-tune calculations (see linked issue)
- [x] Move status from DEVELOPMENT to TESTING

Resolves #3453 
Also resolves #426
closed 2021-02-01 by TD-er