letscontrolit/ESPEasy #430: Rules not

agsochi, 2017-07-11:
if used value from devices like `[sensors#contact] `- nothig published from rules.

`sensors#contact` is `Switch input` gpio 12.

My rules:
```
on system#boot do
 timerSet,1,60
endon

on Rules#Timer=1 do
   Publish %sysname%/sensors/contact,[sensors#contact]
   timerSet,1,60
endon
```
Published only topic name` %sysname%/sensors/contact` without value.
If i try `Publish %sysname%/sensors/contact,1` it published correct.

Grovkillen, 2017-07-11:
Are you using the DEVICE NAME "sensors" and VALUE NAME "contact"? Just to be sure.

agsochi, 2017-07-11:
Yes

Grovkillen, 2017-07-11:
Weird, since the payload is empty that tells me that [sensors#contact] is not set.

on sensors#contact do
   Publish %sysname%/sensors/contact,[sensors#contact]
endon

If you use this rule and change the state of sensors#contact, does the code trigger? Have you monitored the serial log?

agsochi, 2017-07-12:
Yes, state of `sensors#contact` triggered ok, it shows state of water valve.
```
//closed
on sensors#contact=0 do
  //led on
  gpio,15,1
endon

//open
on sensors#contact=1 do
  //led off
  gpio,15,0
endon
```

Grovkillen, 2017-07-12:
If you add a publish command in those snippets of code. Is the [sensors#contact] still empty?

agsochi, 2017-07-12:
Yes, still empty.

Grovkillen, 2017-07-16:
And you have tried the serial logging?

agsochi, 2017-07-17:
No, i use 'mqtt-spy' to debug mqtt messages.

Grovkillen, 2017-07-17:
Ok, but that's only one side of the communication. Please debug serial as well.

SloMusti, 2017-07-28:
You can try the following as well:
 * create a dummy device, single value
 * add into your rule the following ` TaskValueSet,X,1,[sensors#contact]`, where X is the dummy device number on the list
* check the dummy device if it acquired your sensor#contact value

I am using the above functionality with no problem on a rather recent version.

uzi18, 2017-07-28:
Isn't path too long related problem?
Like in sendtohttp issue?

Grovkillen, 2018-10-23:
I just tried this and it seems to not be an issue anymore.

closed 2018-10-23 by Grovkillen