letscontrolit/ESPEasy #432: Plugin categories/type and Plugin states (normal, testing, development) is a bit messy at the moment

Grovkillen, 2017-07-13:
I see some plugins that are on TESTING and DEVELOPMENT are either not noted in the name or have different notes (TEST/TESTING, DEV/DEVELOPMENT). I suggest that we add following to the plugin codes to clean this up and make it more clear:

In plugin.ino:
`#ifdef PLUGIN_BUILD_TESTING

#define PLUGIN_TYPE_XXX      "Type or category" //"Analog", "Environment", "Switch input" etc.
#define PLUGIN_NAME_XXX      "Name of device/circuit"
#define PLUGIN_STATE_XXX      "TESTING" //"NORMAL", "TESTING", "DEVELOPMENT"
`
and...
`    case PLUGIN_GET_DEVICENAME:
      {
        string = F(PLUGIN_TYPE_XXX) + " - " + F(PLUGIN_NAME_XXX);
        #ifdef PLUGIN_BUILD_TESTING
         string = string + " [" + F(PLUGIN_STATE_XXX) + "]";
        #endif
        #ifdef PLUGIN_BUILD_DEV
         string = string + " [" + F(PLUGIN_STATE_XXX) + "]";
        #endif
        break;
      }`

tedenda, 2017-07-13:
👍

Grovkillen, 2017-07-13:
By adding these fields we are able to use them in future releases to add a category drop-down. Imagine having 200 active plugins, that's a long list.

psy0rz, 2017-07-13:
i agree that things should be more consistent. categories might be nice as well. will have to think about a better way to implement it tough.

maybe we should get rid of the whole GET_DEVICE_NAME function and create a cleaner way to let the core handle it.

Grovkillen, 2017-11-21:
I close this.

closed 2017-11-21 by Grovkillen