letscontrolit/ESPEasy #433: Add support for SHT30

budda85, 2017-07-14:
Please, can you add support for the SHT30 Sensor?

THX

jkDesignDE, 2017-07-14:
The plugin should also support the SHT31

beicnet, 2017-07-14:
@jkDesignDE I have it, can you rework it for new releases?, also have done for LM75A...

jkDesignDE, 2017-07-14:
I can do the rework but I have no sensor to test

beicnet, 2017-07-14:
@jkDesignDE I do have them all!

jkDesignDE, 2017-07-14:
@beicnet - Let us write banana-software (maturate at customer)!
I write, you test...
I am busy right now. I can start at sunday.

uzi18, 2017-07-14:
OK

beicnet, 2017-07-14:
@jkDesignDE No problem, just PM me and I will send you my GitHub releases (I take them down for reasons).

budda85, 2017-07-14:
Hey Guys, Thanks for your enthusiasm.
I can test it too. :-)

beicnet, 2017-07-14:
My plugins are working on previous builds to R148 official and to my custom builds to R158, but they are not compatible with current builds.

jkDesignDE, 2017-07-16:
@beicnet - Where can I find the source??? No resporse on PM...

beicnet, 2017-07-16:
@jkDesignDE Bro, you didn't sent me any PM! I did not get nothing! :(

beicnet, 2017-07-17:
@jkDesignDE Did you received my plugin project files regarding SHT3x and LM75?

beicnet, 2017-07-20:
@psy0rz You can close this, it's done! ;)

closed 2017-07-22 by psy0rz

beicnet, 2018-03-05:
@jkDesignDE We have some issue with the [SHT3x](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P068_SHT3x.ino) plugin, can you please verify it as per user [complain](https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=2363&start=20)?

Regards

jkDesignDE, 2018-03-05:
I can have a look at weekend...

beicnet, 2018-03-05:
Thank you! :wink: @jkDesignDE

beicnet, 2018-03-13:
Hey @jkDesignDE did you managed to do something?!

jkDesignDE, 2018-03-14:
@beicnet done but not tested - see #1081 

Same problem for many other plugins!!!

beicnet, 2018-03-14:
@jkDesignDE Thank you, testing it almost right away! :wink:

beicnet, 2018-03-14:
@jkDesignDE How I see, it's reading both sensors now, just to give it a few days continuous test! :wink:

jkDesignDE, 2018-03-15:
A few days continuous test is not necessary because it happens at the first time.
Test: Connect 2 sensors with different I2C addresses. Configure both sensors. Heat up one sensor. Refresh device task page. Reading should be different for both sensors.

beicnet, 2018-03-15:
Yeah, know that, but just in case! Kappa

commit 4da1e4fb (2018-03-15): Fixed #433

Not tested!
commit aab00567 (2018-04-13): Downstream updates (#1)

* Initial re-organisation of plugin sets (#1019)

* 12bit ADC values acceted up to 4095

MCP3221 is a 12bit ADC so accepted values should be up to 40195 not 1024

* automaticly updated release notes for v2.0-20180313

* - move all structs from ESPEasy.ino into header file
- add some fix for headers
- add const to rtttl notes array

* [issue #1066] Fix Webserver handle_control

There wasn't a proper start/end for the streaming webserver.
See discussion on issue #1066

* automaticly updated release notes for mega-20180314

* Custom controller GUI option

* Fixed #433

No [... cut]