letscontrolit/ESPEasy #434: DHT11 sensor on D8 (GPIO-15) no continuous reading

beicnet, 2017-07-14:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Hook up DHT11 sensor to D8 (GPIO-15) pin

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
Tell us what should happen?
Continuous sensor reading

### Actual behavior
Tell us what happens instead?
Dropping reading after each second reading

### System configuration
Only one sensor attached to the board

Hardware:
WeMos D1 mini
DHT11 sensor module with Dupot cables

Software or git version:
ESPEasy_v2.0.0-dev10_normal_4096.bin

[image]

krikk, 2017-07-14:
have a look at this issue #284 ... can you try the fix that is mentioned there and try again...

...all this DHT sensors are really crap...

beicnet, 2017-07-14:
Hey @krikk 

They are not crap, just low cost "accuracy = price" ;)

It's not that issue related, on that pin every second readings are "nan".

Testing all pins now step-by-step, current results are really promising.

Testing at 15 continuous readings:

D0 (GPIO-16)      = Tested /
D3 (GPIO-0)       = Tested /
D4 (GPIO-2)  	  = Tested / OK
D5 (GPIO-14) SCK  = Tested / OK
D6 (GPIO-12) MISO = Tested / OK
D7 (GPIO-13) MOSI = Tested / OK
D8 (GPIO-15) SS   = Tested (not stable?, dropping every second sensor reading)

So, need to test two more pins above...

jkDesignDE, 2017-07-14:
GPIO-15 (D8) has a pulldown resisitor on board (Wemos D1, Node-MCU)! At boot time the pin MUST be LOW!!!
The sensor need a pullup!!! If you use a stonger (less ohms) resistor the sensor should work but the ESP dont boot.
Conclusion: Dont use D8 for DHTxx

beicnet, 2017-07-14:
Thank you @jkDesignDE so, tell me, what GPIOs are with same behavior? Just to avid them!

jkDesignDE, 2017-07-14:
@beicnet - All pins are equal but some pins are more equal than others...

GPIO-15 (D8): Pulldown resisitor 10k; must LOW at boot time
GPIO-0 (D3): Pullup resisitor 10k; must HIGH at boot time
GPIO-2 (D4): Has insternal low-active LED
GPIO-16 (D0): No internal pullup possible; no PWM possible; no interrupt possible
GPIO-1 (D10): TXD1 - Connected to USB-serial converter on board
GPIO-3 (D9): RXD1 - Connected to USB-serial converter on board

beicnet, 2017-07-14:
Thank you! @jkDesignDE

psy0rz, 2017-07-16:
@jkDesignDE  more details on the io pins and their usability are here: https://www.letscontrolit.com/wiki/index.php/Configuration

closed 2017-07-16 by psy0rz