letscontrolit/ESPEasy #435: MQTT do not work

vr6syncro, 2017-07-14:
### Steps to reproduce
How can we trigger this problem?

Simple connect to a MQTT 
MQTT : Connection lost
MQTT : Failed to connected to broker
 test.mosquitto.org or local or other broker did not work

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes

Hardware:
Wemos and NodeMCU

Software or git version:
148 are working well. all 200 version are not working for me.

Grovkillen, 2017-07-15:
Take this to the forum. I use multiple 2.0+ versions, no problem to connect to my broker (local IP or global URL).

vr6syncro, 2017-07-15:
can u test if test.mosquitto.org working for u?

Grovkillen, 2017-07-15:
It works.

vr6syncro, 2017-07-15:
thats strange

i am using stock firmware nothing changed same setting as fw 148

Grovkillen, 2017-07-15:
Have you reentered the settings? I can confirm that my test node using 2.0.0-dev10 connect all right.

vr6syncro, 2017-07-15:
i format and reinstalled the normal version now its working. 
i try dev again with the next release

Grovkillen, 2017-07-15:
Good to know, please close this issue.

infernix, 2017-07-18:
I'd like to reopen this - I upgraded mosquitto from 1.3.x to 1.4.12 and am seeing the exact same thing, both by anonymous and username/password access.

@Grovkillen what version did you test against?

Grovkillen, 2017-07-18:
1.3 i would think.

infernix, 2017-07-18:
On 1.4 I am seeing:
```
1500403131: Socket error on client <unknown>, disconnecting.
1500403131: New connection from 192.168.0.109 on port 1883.
1500403131: Socket error on client <unknown>, disconnecting.
1500403135: New connection from 192.168.0.115 on port 1883.
1500403135: Socket error on client <unknown>, disconnecting.
1500403135: New connection from 192.168.0.115 on port 1883.
```
WIll try to debug further

infernix, 2017-07-18:
Seems to be an issue with anonymous connects; once i switched to authenticated things started working again.

infernix, 2017-07-18:
I had half a dozen ESPEasys running for months before i upgraded to 1.4.12 so pretty sure there's an issue there, unfortunately too busy to dig deeper at this point since i wanted to switch to authentication anyway :)

closed 2017-09-17 by vr6syncro