letscontrolit/ESPEasy #437: No IDX option when using "Output - (Domoticz MQTT Helper)"

22hn, 2017-07-16:
When setting up a device as "Output - (Domoticz MQTT Helper)" there no option to configure the Domoticz IDX value.

### System configuration
Hardware: Wemos D1 mini
Software or git version: v2.0.0-dev10 normal

psy0rz, 2017-07-16:
the idx number is configured per device, not per controller. so this setting is on the device page.

closed 2017-07-16 by psy0rz

22hn, 2017-07-16:
That was my point...! Where to put the IDX?

>>> DatuX <notifications@github.com> 2017-07-16 14:55 >>>
the idx number is configured per device, not per controller. so this
setting is on the device page.
—
(
https://github.com/letscontrolit/ESPEasy/issues/437#issuecomment-315607490)
(

psy0rz, 2017-07-16:
just add a device, and there you can input the idx.

22hn, 2017-07-16:
In 1.20 I can... In 2.0 I can not...

Did you look at my screenshot from last email?

Hakan

>>> DatuX <notifications@github.com> 2017-07-16 15:41 >>>
just add a device, and there you can input the idx.
—
(
https://github.com/letscontrolit/ESPEasy/issues/437#issuecomment-315609814)
(

22hn, 2017-07-16:
Could be this issue...

https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3041&p=16713&hilit=idx+missing#p16713

>>> DatuX <notifications@github.com> 2017-07-16 15:41 >>>
just add a device, and there you can input the idx.
—
(
https://github.com/letscontrolit/ESPEasy/issues/437#issuecomment-315609814)
(

psy0rz, 2017-07-16:
i dont see the screen-shots since you're mailing to github. please add the screenshot here: 

https://github.com/letscontrolit/ESPEasy/issues/437#issuecomment-315609981

psy0rz, 2017-07-16:
ah that bug might be fixed already, try with the latest version from github. (not dev10)

22hn, 2017-07-16:
Hi!

Unfortunately, I don't have the knowlegde (yet) to compile myself...
That's why I went with the last published release...

The reason I started looking at version 2 was that I couldn't get the
mqtt output to work in 1.20 either. 

Read something in the forum thread about mqtt messages could only be
256 bytes of length...? I can see that the mqtt message coming from
Domoticz is 285 bytes in this case. Could that be the reason and in that
case, how do I fix it...?

Hakan

>>> DatuX <notifications@github.com> 2017-07-16 16:45 >>>
ah that bug might be fixed already, try with the latest version from
github. (not dev10)
—
(
https://github.com/letscontrolit/ESPEasy/issues/437#issuecomment-315613761)
(

psy0rz, 2017-07-16:
i dont know. maybe someone else knows. i'm planning to release dev11 and to close most of the open 2.0.0 issues.

22hn, 2017-07-16:
OK, sounds great!

In the progress of setting up Platformio now, we'll see how that
goes... :-)

>>> DatuX <notifications@github.com> 2017-07-16 17:12 >>>
i dont know. maybe someone else knows. i'm planning to release dev11
and to close most of the open 2.0.0 issues.
—
(
https://github.com/letscontrolit/ESPEasy/issues/437#issuecomment-315615692)
(

Grovkillen, 2017-07-17:
Just as a notice, max size of MQTT messages is NOT the issue. The standard states that max size (header, QoS, topic, payload etc) of 268,435,455 (256 MB).

Reference: http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html
(2.2.3 Remaining Length)

psy0rz, 2017-07-17:
@22hn just released v2.0.0-dev11. check it out :)