letscontrolit/ESPEasy #440: I2C initialized after rules "system boot

manjh, 2017-07-18:
When using a Sonoff 4CH, two of the relays are linked to GPIO4 end 5, which are normally used for SDA/SCL. 
In the 4CH I needed to unlink these GPIO's on the HW page. Not a problem.
The 4CH seems fully operational, except for one problem.
I use the "System#Boot"  event in the rules section to initialize some relays after system boot. 
When I switch GPIO5 and 4 on, these relays go into a very strange state: toggle command (also in a set of rules) no longer have any effect. Only after an explicit "off" command from Domoticz, will the relays respond OK again.

On a tip from Grovkillen I made a workaround: in the system boot section I started a timer with a delay of 5 seconds. When the timer runs out I then have some rules to init the relays.
Using this method as a workaround makes the 4CH work OK.
But there is something wrong here. It is suspected that the I2C initialization happens AFTER the system boot section in the rules, which is not as it should be. 
The rules are controlled by the user, and should run only AFTER complete system init.

mvdbro, 2017-09-23:
Current boot sequence (simplified):

1)  hardwareInit();
    set gpio pin defaults
    if (Settings.Pin_i2c_sda != -1)
      Wire.begin

2)  PluginInit();

3)  String event = F("System#Boot");
    rulesProcessing(event);

The System#Boot event is the last action within the Setup()

Must be some other things going on here...

manjh, 2017-09-23:
Tsja, je zou zeggen dat het goed moet gaan, maar..

Hans

psy0rz, 2017-09-23:
@manjh did you try this with dev11/12? i might have changed that order after Jul 18.

manjh, 2017-09-24:
I don't understand. Have I tried what?

psy0rz, 2017-09-26:
can you try it with version 2.0.0-dev12 of espeasy, to see if it fixed.

manjh, 2017-09-26:
Try it to fix what, exactly? What was the problem?

Hans

Grovkillen, 2017-10-10:
@manjh read your own post and you'll see that the problem you had was "dirty fixed" using a delay. Psy0rz want you to try that again after updated to DEV12.

manjh, 2017-10-10:
The problem is that none of the mails on this subject contains a link to
the original problem, or even a short description! I have posted a lot of
things over time.... which one is it?

Hans

Grovkillen, 2017-10-10:
> 440#issuecomment-335377910

Further down in this email (you might need click to show more in the bottom
of the email). You'll see this link:
https://github.com/letscontrolit/ESPEasy/issues/440#issuecomment-335409229
but with the link text " view it on GitHub
<https://github.com/letscontrolit/ESPEasy/issues/440#issuecomment-335409229> ",
click that one.

You'll also see the name of the issue on the subject of this email:  I2C
initialized after rules "system boot (#440)

You could also use this #440 number to search on GitHub.

They all work just fine.

psy0rz, 2017-11-16:
manjh het gaat om deze issue: https://github.com/letscontrolit/ESPEasy/issues/440

met welke versie heb je dit getest? voor zover we kunnen zien klopt de initalisatie volgorde nu.

closed 2017-11-16 by psy0rz

manjh, 2017-11-16:
Ik ben dit weekend weer thuis, zal proberen of ik het even kan testen.

Hans

manjh, 2017-11-20:
At this moment I am unable to test this fix. Both my units are built in,
and in order to test I would need to take them out and open them up for
installation. Current build is B147, and the new build would require
connection via serial in order to get the units to connect (log in to
Wifi).

Hans

psy0rz, 2017-11-20:
Ok try it in the future, if its still a problem please let us know. :)

manjh, 2017-11-20:
The problem is not only the fact that I have to take the unit out of where
it is now....
1. remove the unit from where it is built in
2. open the Sonoff enclosure
3. connect to computer and load
4. enter all device definitions, rules, and settings
5. test

Depending on test results: go back to current build.
And then, of course, steps 1 and 2 in reversed order...

It might be worth getting an extra Sonoff 4CH and try it on that. Will
come in handy in the future to have a spare. Will think about it.