letscontrolit/ESPEasy #441: Improved PWM support

infernix, 2017-07-18:
The SDK PWM is a bit limited:

- Duty cycle limited to 90% (at 1kHz PWM period)
- usable PWM period at most ~2KHz.
- Incomplete documentation

There is a better implementation available here: https://github.com/StefanBruens/ESP8266_new_pwm (GPLv2):

> This replacement allows duty cycles from 0% to 100%, with a stepsize of 200ns. This is 5000 steps for a 1kHz PWM, and 256 steps (8 bit of resolution) at 19kHz.

With one caveat of which I'm not sure is going to be a problem:

> The code uses the TIMER1 interrupt. If you use e.g. the softtimer, there is a conflict. You can use NM1 for the PWM instead.

Is there any interest to adopt this as a replacement or as an alternative to the existing PWM functionality (which does not work for my case)? If so I can look into making a PR for it.

jkDesignDE, 2017-07-18:
It would be nice to have flicker-free and full functional PWM on ESP.
In this lib I am missing an arbitrary mapping between GPIO numbers and the MUX and FUNC values. I see also a problem in adding and removing pins on the fly (clicking in web interface)
May be this PWM can be implemented in a separate plugin (additional to the internal PWM)

buyapeng, 2017-09-25:
How can I use PWM send IR Code ?
i want receive IR code from Mqtt and PWM send.
on sniff#cmdMQTT=2 do
  PWM,2,<?>
endon

sassod, 2017-09-28:
@Gerben

I would really like to see better pwm handling especially the one that
doesen't use delays in the implementation. Currently as I understand the
pwm delays all other functions until completion.

Sincerly
David

mikkel75, 2018-12-11:
Is it possible to test this new PWM in a beta build?

Grovkillen, 2019-10-11:
Ok, lets open it.

tonhuisman, 2023-09-03:
IR sending (and receiving) has a dedicated plugin, with command support, so this issue is no longer relevant, and can be closed.

closed 2023-09-03 by TD-er

xnrfn, 2023-09-05:
For anyone reading this in the future I have no idea what IR send/receive has to do with improved PWM support - this issue was opened for sending a better PWM signal on GPIOs, not for anything IR related.

But since this hasn't moved since 2017, it's probably best left closed.

infernix, 2023-09-05:
For anyone reading this in the future I have no idea what IR send/receive has to do with improved PWM support - this issue was opened for sending a better PWM signal on GPIOs, not for anything IR related.

But since this hasn't moved since 2017, it's probably best left closed.

tonhuisman, 2023-09-05:
Have you tried any recent release of ESPEasy? As a lot has changed since 2017, including newer versions of the Arduino libraries, and tough I don't have detailed info on the base PWM behavior, it is very likely that it has improved, over time, and also controlling the PWM output has improved in ESPEasy.