letscontrolit/ESPEasy #443: P063 "KeyPad - TTP229 Touch [TESTING]"

Grovkillen, 2017-07-18:
I have now tested this plugin and it works just great, except that it only takes the first 8 keys... I have a 16 key board (see link below) and the 9-16 keys are just ignored by the plugin.

[TTP229](https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042311.0.0.elxtvC&orderId=503052459820475&productId=2055105031)

jkDesignDE, 2017-07-18:
@Grovkillen - You have to solder a bridge on the board!

From manual:
Input keys number select - The  TTP229-LSF  has  8  keys  input  mode  and  16 keys input mode. hese modes are selected via high-value resistor connected to the TP2(KYSEL) pin to VSS, or not. The efault hat TP2(KYSEL) pin is not used resistor connected to VSS is selected 8 keys input mode. Another is elected 6 keys input mode that has used a high-value resistor connected to VSS.

Grovkillen, 2017-07-18:
I got it working now. Sorry! But the i2c scanner issue is real?

jkDesignDE, 2017-07-18:
mportant: There are several types of TTP299 chips with different features available. They are namead all TTP229 but differ in the letter(s) followed.
On the china boards (found on eBay and AliExpress) the TTP229-B is used which has NO! I2C-interface. It uses a properitary serial protocol with clock (SCL) and bidirectional data (SDO)

DO NOT USE THE I2C PINS!!!

Grovkillen, 2017-07-18:
So this should be stated on the wiki. Should we never use i2c pins or what do you mean?

jkDesignDE, 2017-07-18:
If you connect this device to the I2C pins the (real) I2C devices are malfunctioning

Grovkillen, 2017-07-18:
Ah, great to know! I'm writing the wiki as we talk...

Grovkillen, 2017-07-18:
@jkDesignDE is this something you feel is correct: https://www.letscontrolit.com/wiki/index.php/TTP229

jkDesignDE, 2017-07-18:
@Grovkillen - Wrong description!
The module you have is a TTP229-B and the plugin is exactly for this TTP229-B! No I2C!
The pin is SDO not SDA (as on I2C devices).
Use unly 3.3V - 5V will damage the ESP.
In multi-key-mode he codes are the sum of all pressed codes.

Grovkillen, 2017-07-18:
@jkDesignDE 👍 Thanks for quick feedback.

Grovkillen, 2017-07-18:
I have now (hopefully) corrected to your satisfaction :)

Grovkillen, 2017-07-19:
@jkDesignDE I will close this since I was misunderstanding some basic stuff. The Wiki page is done accordingly.

closed 2017-07-19 by Grovkillen