letscontrolit/ESPEasy #446: RTTTL over MQTT in "endless" loop

Grovkillen, 2017-07-19:
If I start a jingle using this command:

Topic: ESP_Easy/cmd
Payload: rtttl,14,d=4,o=5,b=112:8a,8a,a,8a,8a,a,8a,8c6,8f.,16g,2a,8a#,8a#,8a#.,16a#,8a#,8a,8a.,16a,8a,8g,8g,8a,g,c6

And disconnect the MQTT broker whilst the tune is playing the tune will start play again for each **MQTT Broker: Connection lost** and **MQTT Broker: Connected**.

And the ESP_Easy/status doesn't ever publish the json message

```
{
"log": "rtttl,14:d=4,o=5,b=112:8a,8a,a,8a,8a,a,8a,8c6,8f.,16g,2a,8a#,8a#,8a#.,16a#,8a#,8a,8a.,16a,8a,8g,8g,8a,g,c6",
"plugin": 1,
"pin": 14,
"mode": "output",
"state": 0
}
```

psy0rz, 2017-07-19:
can you try with a shorter simpler tune. the rtttl code is very buggy: the slightest mistake in input and it will crash or corrupt memory :P

Grovkillen, 2017-07-19:
Will try...! :)

Grovkillen, 2017-07-19:
@psy0rz The tune: `d=10,o=6,b=180,c,e,g` is so short I cannot disconnect the broker to test this bug. Any suggestions?

Grovkillen, 2017-07-19:
https://www.letscontrolit.com/wiki/index.php/Buzzer_(RTTL)

I've updated the plugin page, please check and see. 🥇

Grovkillen, 2018-03-12:
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=4985&start=20#p25841

Okay, we have some more information in the matter.

uzi18, 2018-03-12:
if you want to use 8a# replace it first to 8a- as before push to rtttl command all - are replaced to #

Grovkillen, 2018-03-12:
But could it be allowed to use # in the tunes? Since it's commonly used in RTTTL melodies...

uzi18, 2018-03-12:
Dont know why such modification is used, but we know there is a bug somewhere in playing when use # char.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er