letscontrolit/ESPEasy #449: Custom build (Platformio) does not make GIT_BUILD="custom"

Grovkillen, 2017-07-22:
```
#ifndef BUILD_GIT
#define BUILD_GIT "(custom)"
#endif
```
The above code seems to not do anything for me since the "GIT version:" is blank...

psy0rz, 2017-07-22:
its a bit difficult to do this in a generic way. it would be awesome if "git describe" output could be there in some way, but i dont think there is a clean way for it that works in mac/linux/windows at the same time.

Grovkillen, 2017-07-22:
But the code snippet I posted is not working?

Grovkillen, 2017-09-15:
@psy0rz cant we make a dirty fix as posted by me? I just want the build to be populated with "custom" if the build is self-compiled.

psy0rz, 2017-09-16:
yep i think so. i will try if it works in all situations

psy0rz, 2017-10-12:
This is actually surprisingly hard to fix in a nice way.  I spend like 45 minutes on it. (Your solution is also invalid, since it always defined it as "(custom)")

So you're out of luck ;)

closed 2017-10-12 by psy0rz

Grovkillen, 2017-10-13:
Ah, too bad. Thanks for nocking your head into wall. :)