letscontrolit/ESPEasy #450: GPIO vs D0...D12

Grovkillen, 2017-07-23:
This is a general thought. Since we have both GPIO and Dx descriptions in the web interface it's thought that you would be able to use the Dx descriptions for cmd.

These would be the same:
```
GPIO,14,1
GPIO,D5,1
```

Is there anything stopping this idea?

psy0rz, 2017-09-13:
this sounds like a good idea, that needs a generic function to make the translations from DX -> gpio number.

something for the 2.1 version though.

TD-er, 2018-12-06:
These Dx labels are not useful on ESP32.
Also they are mainly used on NodeMCU boards and the Wemos D1 mini used them too.
Not sure if they make any sense on other boards.

Grovkillen, 2018-12-06:
I know but we will have ESP82XX units alive for some years and they are labeled that way. Very bad decision by the designer but still, they will be referred to as Dx.

giig1967g, 2018-12-06:
I think it's a good idea.

TD-er, 2019-10-10:
Is this still a good idea?
One reason not to do it, may be that a parser for these commands must allow a string instead of an integer value.
Not sure if we have multiple implementations for such parsers.
Anyway if we're going to add this, we need to create a separate "gpio pin parser" function which is being called from everywhere where a GPIO pin is expected.

Do we also have places where we output a GPIO number? Should that also be using the NodeMCU labeling?

Grovkillen, 2019-10-10:
Nah, the ESP32 uses GPIO numbers and it would not be a good idea of messing with two different ways of addressing the pins. I close.

closed 2019-10-10 by Grovkillen