letscontrolit/ESPEasy #452: Counter plugin vs GPIO16

ShardanX, 2017-07-25:
### Steps to reproduce
Using a pulse counter on GPIO16

Does the problem presist after powering off and on? 
Yes

### Expected behavior
Counting pulses

### Actual behavior
Does not count

### System configuration

Hardware: ESP12-E

Software or git version:
Tested with R148 and 2.0.0 from July (selfcompiled)

I've tested this configuration - the counter does not work on GPIO16.
As GPIO16 cannot use interrupts this pin might be useless for counters.

This should be mentioned within Wiki, or the GPIO16 should be greyed out
in the selection pulldown menu.

Grovkillen, 2017-07-25:
Or it might be enough to have a note (like with GPIO 9) stating this for the pulse counter plugin?

jkDesignDE, 2017-07-26:
Same for P008, P013, P046, P059

Grovkillen, 2017-10-06:
A simple statement of this on the plugin pages should be enough.

psy0rz, 2017-10-09:
indeed, can you add that @Grovkillen ? and then close this.

Grovkillen, 2017-10-10:
I will do that @psy0rz give me a day :)

Grovkillen, 2017-10-10:
I have now updated the following pages:

https://www.letscontrolit.com/wiki/index.php/Configuration
https://www.letscontrolit.com/wiki/index.php/LJ12A3
https://www.letscontrolit.com/wiki/index.php/Rotary_Encoder
https://www.letscontrolit.com/wiki/index.php/VentusW266
https://www.letscontrolit.com/wiki/index.php/Wiegand-26

closed 2017-10-10 by Grovkillen

psy0rz, 2017-10-12:
thanks! you're the best:)