letscontrolit/ESPEasy #454: Counter - different outputs on device page and display

ShardanX, 2017-07-25:
### Steps to reproduce
Configuring counter to "Delta" and "Delta/Total"
Show readings on device page and a connected display.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes
### Expected behavior
Results should be same between display and device page.

### Actual behavior
On device page i see for example Count: 47 and Total: 1037, no matter if i configure
"Delta" or "Delta/Total" mode.
If set to "Delta" these values are shown on a connected OLED display as in device page.

If i change the setting to "Delta/Total" both lines in the display show the "Total" value.
On the device page both values are still different.

Display settings are ok, otherwise with "Delta" it would show same value too.

### System configuration

Hardware: 
ESP-12-E, Frequency generator to simulate input, OLED display

Software or git version: 2.0.0 custom (sources 08. July 2017)

MatzeJoerling, 2017-08-30:
i also have some issues with pulse counter. cant really track it exactly down, but since update to dev9/dev10 or dev11 (dont know) is show not / sends not the correct values.

Grovkillen, 2018-10-25:
Is this still an issue @ShardanX ? If so please open it.

closed 2018-10-25 by Grovkillen