letscontrolit/ESPEasy #456: Rules: sendtohttp cmd not saved correctly

uzi18, 2017-07-26:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Put line longer then 64 chars,  after save rules it will reload with malformed line. 

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
Save correct line.
Usually url is not so short. 

### Actual behavior
Line is splitted in the middle,  some chars are lost. 

### System configuration

Hardware:

Software or git version:

Grovkillen, 2018-03-01:
@TD-er is this something we can make for 2.0.0?

TD-er, 2018-03-01:
Just to be sure, this line is just a line in the rules text field?
Can you add a screenshot (and cut and paste of the text) to show it?

uzi18, 2018-03-01:
Hah,  don't remember now,  just put long line in rules textbox and save.
Refresh rules page,  is this line same long like before save?

TD-er, 2018-03-01:
I just thought the rules were just a text field up to 2048 bytes.
Never thought it would be stored as separate lines, I thought the line ending would just be something like /n/r

uzi18, 2018-03-01:
After about 1h will try reinvestigate it.

uzi18, 2018-03-01:
Can't reproduce it now. 
Maybe fixes for webserver resolved this problem.

closed 2018-03-01 by uzi18