letscontrolit/ESPEasy #459: Serial Server wont work

DittelHome, 2017-07-30:
Hi,
i try to get serial connection from FHEM to a serial homeautomation adapter (HMUART)
I am using this device:
[]http://www.raspifun.de/viewtopic.php?t=32

I use TX and RX pins. The lines are crossed. The device is connected with four pins, 3,3 V, GND, TX, RX.
Baudrate, start and stopbit, and parity is set correct.

I only get this messages: 

> 2467348 : WD : Uptime 41 ConnectFailures 0 FreeMem 22352
> 2481913 : Ser2N: Client connected!
> 2495008 : Ser2N: Client connected!
> 2497349 : WD : Uptime 42 ConnectFailures 0 FreeMem 22144
> 2508115 : Ser2N: Client disconnected!
> 2508216 : Ser2N: Client connected!
> 2521165 : Ser2N: Client disconnected!
> 2521266 : Ser2N: Client connected!

The server dont get a response from my WEMOS mini D1.
I get no answer !

At server side i get only the following messages: 
> 2017.07.29 00:18:11 3: HMUART2 device closed
> 2017.07.29 00:18:11 1: 192.168.1.28:23 reappeared (HMUART2)
> 2017.07.29 00:18:15 1: HMUARTLGW HMUART2 did not respond for the 1. time, resending
> 2017.07.29 00:18:18 1: HMUARTLGW HMUART2 did not respond for the 2. time, resending
> 2017.07.29 00:18:21 1: HMUARTLGW HMUART2 did not respond for the 3. time, resending
> 2017.07.29 00:18:24 1: HMUARTLGW HMUART2 did not respond after all, reopening
> 2017.07.29 00:18:24 3: HMUART2 device closed
> 2017.07.29 00:18:24 1: 192.168.1.28:23 reappeared (HMUART2)
> 2017.07.29 00:18:28 1: HMUARTLGW HMUART2 did not respond for the 1. time, resending
> 2017.07.29 00:18:31 1: HMUARTLGW HMUART2 did not respond for the 2. time, resending
> 2017.07.29 00:18:34 1: HMUARTLGW HMUART2 did not respond for the 3. time, resending
> 2017.07.29 00:18:37 1: HMUARTLGW HMUART2 did not respond after all, reopening
> 2017.07.29 00:18:37 3: HMUART2 device closed

With esp-link the HMUART is running, but i need ESPEASY because i want to connect some peripherals...
Under advanced the serial port is enabled.
Using Dev11...

regards
Klaus

immiimmi, 2017-07-30:
Hi klaus
I agree with you, ser2net is not functioning properly anymore.
I merged some old code, now it is working for me.
I am sure the experts can do better.
immi
[_P020_Ser2Net.ino.zip](https://github.com/letscontrolit/ESPEasy/files/1185698/_P020_Ser2Net.ino.zip)

DittelHome, 2017-07-30:
Hi Immi,
thanks for the information. I have tested your version, but for me, same situation.
I believe that the ser2net doesnt support RAW data over serial...
The first sequence which is sent to the adapter is: 00 00
After then, i get no response.
Klaus

DittelHome, 2017-07-30:
This are the strings which will be sent normally:

> 2017.07.30 19:31:10 3: HMUART1 device opened
> 2017.07.30 19:31:11 0: HMUARTLGW HMUART1 send: 00 00
> 2017.07.30 19:31:11 0: HMUARTLGW HMUART1 recv: 00 0402436F5F4350555F424C, state 1
> 2017.07.30 19:31:11 0: HMUARTLGW HMUART1 currently running Co_CPU_BL
> 2017.07.30 19:31:11 0: HMUARTLGW HMUART1 send: 00 03
> 2017.07.30 19:31:11 0: HMUARTLGW HMUART1 recv: 00 0401, state 2
> 2017.07.30 19:31:11 0: HMUARTLGW HMUART1 recv: 00 00436F5F4350555F417070, state 2
> 2017.07.30 19:31:11 0: HMUARTLGW HMUART1 currently running Co_CPU_App
> 2017.07.30 19:31:12 0: HMUARTLGW HMUART1 send: 01 0029A083
> 2017.07.30 19:31:12 0: HMUARTLGW HMUART1 recv: 01 0401, state 4
> 2017.07.30 19:31:12 0: HMUARTLGW HMUART1 GetSet Ack: 01, state 4
> 2017.07.30 19:31:12 0: HMUARTLGW HMUART1 send: 01 01
> 2017.07.30 19:31:12 0: HMUARTLGW HMUART1 recv: 01 0407010129A083, state 5
> 2017.07.30 19:31:12 0: HMUARTLGW HMUART1 GetSet Ack: 07, state 5
> 2017.07.30 19:31:12 0: HMUARTLGW HMUART1 send: 01 10
> 2017.07.30 19:31:12 0: HMUARTLGW HMUART1 recv: 01 0407010145F973, state 6
> 2017.07.30 19:31:12 0: HMUARTLGW HMUART1 GetSet Ack: 07, state 6
> 2017.07.30 19:31:13 0: HMUARTLGW HMUART1 send: 00 0E597E17E104
> 2017.07.30 19:31:13 0: HMUARTLGW HMUART1 recv: 00 0401, state 7
> 2017.07.30 19:31:13 0: HMUARTLGW HMUART1 GetSet Ack: 01, state 7
> 2017.07.30 19:31:13 0: HMUARTLGW HMUART1 send: 00 02
> 2017.07.30 19:31:13 0: HMUARTLGW HMUART1 recv: 00 0402010003010401, state 8
> 2017.07.30 19:31:13 0: HMUARTLGW HMUART1 GetSet Ack: 02, state 8
> 2017.07.30 19:31:13 0: HMUARTLGW HMUART1 send: 00 06
> 2017.07.30 19:31:13 0: HMUARTLGW HMUART1 recv: 00 0401, state 10
> 2017.07.30 19:31:13 0: HMUARTLGW HMUART1 GetSet Ack: 01, state 10
> 2017.07.30 19:31:13 0: HMUARTLGW HMUART1 send: 00 0B

immiimmi, 2017-07-30:
hi klaus
happy that you see an improvement; now you should try to understand, when is working and when it is not. ciao immi

DittelHome, 2017-08-04:
If i use this plugin (other wiring neccessary) the communication to the serial device works....
[_P130_SSSRV..ino.txt](https://github.com/letscontrolit/ESPEasy/files/1200623/_P130_SSSRV.ino.txt)
...something must wrong in the serial server ...

DittelHome, 2017-08-16:
Problem is now solved !
In the TX/RX line of the uart adapter was a 4k7 resistor in line, After removing both resistors the Serial Server works fine.
In addition all log levels must be set to 0 and the serial port must be enabled.

closed 2017-08-16 by DittelHome

tavoc, 2020-05-15:
> 
> 
> Problem is now solved !
> In the TX/RX line of the uart adapter was a 4k7 resistor in line, After removing both resistors the Serial Server works fine.
> In addition all log levels must be set to 0 and the serial port must be enabled.

Thanks. This should be mentioned somewhere. I searched days why its working with an old rpi and not a nodemcu

TD-er, 2020-05-15:
Hmm where should these resistors be located?
As far as I know the NodeMCU (and others) only have a 500 (not a 470, but 500) Ohm resistor in series with the RX/TX line to prevent 'ringing' as is also described in the ESP8266 and ESP32 data sheets.

tavoc, 2020-05-15:
> 
> 
> Hmm where should these resistors be located?
> As far as I know the NodeMCU (and others) only have a 500 (not a 470, but 500) Ohm resistor in series with the RX/TX line to prevent 'ringing' as is also described in the ESP8266 and ESP32 data sheets.

Its not the resistor on the nodemcu, but on the Homematic Board HM-MOD-RPI-PCB