letscontrolit/ESPEasy #460: SendToHTTP issue

berbergh, 2017-07-30:
I use a script to switch a light on Domoticz:
on E16Nextion#code=7 do
    if [E16Nextion#value]=1
        SendToHTTP 192.168.0.243,8080,/json.htm?type=command&param=switchlight&idx=7&switchcmd=On
    else
        SendToHTTP 192.168.0.243,8080,/json.htm?type=command&param=switchlight&idx=7&switchcmd=Off
    endif
endon

This works fine, but not when assembled with #define PLUGIN_BUILD_TESTING

I expect the light to be switch, but when ik development, it does not.

I use a WeMos D1 mini, the firmware is copied from the github and compiled with Arduino 1.6.11

I have tried the versions of ESPEasy_v2.0.0-dev4 to 11

psy0rz, 2017-08-02:
thats strange...can you increase the serial logging and copy paste the output here when its doing the send?

berbergh, 2017-08-02:
This is with the PLUGIN_BUILD_NORMAL:

Nextion : code: ,,
EVENT: E16Nextion#code=773.00
ACT  : Nextion,tm0.tim=20000
tm0.tim=20000
EVENT: Processing time:158 milliSeconds
EVENT: E16Nextion#value=1.00
EVENT: Processing time:119 milliSeconds
HTTP : connecting to 192.168.0.243:8080
HTTP/1.1 200 OK

HTTP : Success
Content-Length: 53

Content-Type: application/json;charset=UTF-8

Cache-Control: no-cache

Pragma: no-cache

Access-Control-Allow-Origin: *

{
   "status" : "OK",
   "title" : "Update Device"
}
HTTP : closing connection
Nextion : send command: |s,i7,sOff!²7.00
EVENT: E16Nextion#code=7.00
ACT  : Nextion,tm0.tim=20000
tm0.tim=20000
ACT  : SendToHTTP 192.168.0.243,8080,/json.htm?type=command&param=switchlight&idx=7&switchcmd=Off
HTTP/1.1 200 OK

EVENT: Processing time:185 milliSeconds
EVENT: E16Nextion#value=0.00
EVENT: Processing time:121 milliSeconds
HTTP : connecting to 192.168.0.243:8080
HTTP/1.1 200 OK

HTTP : Success
Content-Length: 53

Content-Type: application/json;charset=UTF-8

Cache-Control: no-cache

Pragma: no-cache

Access-Control-Allow-Origin: *

{
   "status" : "OK",
   "title" : "Update Device"
}
HTTP : closing connection

End below with PLUGIN_BUIL_TESTING:

Nextion : code: ,,
EVENT: E16Nextion#code=773.00
ACT  : Nextion,tm0.tim=20000
tm0.tim=20000
EVENT: Processing time:183 milliSeconds
EVENT: E16Nextion#value=1.00
EVENT: Processing time:118 milliSeconds
HTTP : connecting to 192.168.0.243:8080
HTTP/1.1 200 OK

HTTP : Success
Content-Length: 53

Content-Type: application/json;charset=UTF-8

Cache-Control: no-cache

Pragma: no-cache

Access-Control-Allow-Origin: *

{
   "status" : "OK",
   "title" : "Update Device"
}
HTTP : closing connection
Nextion : send command: |s,i7,sOff!²7.00
EVENT: E16Nextion#code=7.00
ACT  : Nextion,tm0.tim=20000
tm0.tim=20000
ACT  : SendToHTTP 192.168.0.243,8080,/json.htm?type=command&param=switchlight&idx=7&switchcmd=Off
HTTP/1.1 200 OK

EVENT: Processing time:185 milliSeconds
EVENT: E16Nextion#value=0.00
EVENT: Processing time:119 milliSeconds
HTTP : connecting to 192.168.0.243:8080
HTTP/1.1 200 OK

HTTP : Success
Content-Length: 53

Content-Type: application/json;charset=UTF-8

Cache-Control: no-cache

Pragma: no-cache

Access-Control-Allow-Origin: *

{
   "status" : "OK",
   "title" : "Update Device"
}
HTTP : closing connection

psy0rz, 2017-09-13:
Can you change DEVICES_MAX to 64 in ESPEasy.ino, and try again?
`#define DEVICES_MAX                      64`

berbergh, 2017-09-14:
It did not help. Instead, via the serial connection, I got the warning: System error, too much plugins.

Verzonden vanuit Mail voor Windows 10

Van: DatuX
Verzonden: woensdag 13 september 2017 19:13
Aan: letscontrolit/ESPEasy
CC: berbergh; Author
Onderwerp: Re: [letscontrolit/ESPEasy] SendToHTTP issue (#460)

Can you change DEVICES_MAX to 64 in ESPEasy.ino, and try again?
#define DEVICES_MAX 64
—

---
Deze e-mail is gecontroleerd op virussen door AVG.
http://www.avg.com

psy0rz, 2017-09-14:
oh right, delete some plugins as well and try again please.

berbergh, 2017-09-14:
Sorry,
Did not help. I mean, the error is gone, but the light is still not switched.
Just for the record, I am enabling:
//build all plugins that are in test stadium
//#define PLUGIN_BUILD_TESTING

Verzonden vanuit Mail voor Windows 10

Van: DatuX
Verzonden: donderdag 14 september 2017 20:30
Aan: letscontrolit/ESPEasy
CC: berbergh; Author
Onderwerp: Re: [letscontrolit/ESPEasy] SendToHTTP issue (#460)

oh right, delete some plugins as well and try again please.
—

---
Deze e-mail is gecontroleerd op virussen door AVG.
http://www.avg.com

psy0rz, 2017-11-16:
i have no idea what it could be. 

which plugin is generating the E16Nextion#code event?

psy0rz, 2017-11-16:
ah i can reproduce it. (i used my own domoticz and the system#boot event)

its the dreaded lower/upercase bug, will fix that one first

psy0rz, 2017-11-16:
fixed in that last commit.

closed 2017-11-16 by psy0rz