letscontrolit/ESPEasy #461: value names get incorrectly mangled when sent via HTTP Advanced

RREE, 2017-07-31:
### Steps to reproduce
Define a controler "Generic HTTP Advanced" with POST method and a body like %1%%vname1% value=%val1%%/1%

If the value name contains an underscore (e.g. T_Dach) the underscore gets escaped for http transport into %5f and the resulting message from the above definition is T%5fDach instead of T_Dach

Does the problem persist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
No escaping of special characters that are part of the value name

### Actual behavior
The name T_Dach get incorrectly mangled into T%5fDach

### System configuration
Wemos mini D1 with ESPEasy version v2.0.0-dev11

Hardware:
Wemos mini D1 

Software or git version:
v2.0.0-dev11

krikk, 2017-08-03:
according to [wikipedia](https://en.wikipedia.org/w/index.php?title=POST_(HTTP)&action=edit&section=2) ...the default is to url encode the body of the post request, so the current behaviour seems ok for me... 

post-ing without url-encoding is only defined if sending a content type of "multipart/form-data" ...so disabling url encoding for values can only be optional and would need another option in plugins settings, not really sure if we want to do this for this not very common use-case...

RREE, 2017-08-05:
According to the standards it is probably OK to encode the content of a POST message. The problem is that I want to use it to send the values to an Influx DB. Influx expects (as far as I can practically see) unencoded strings, So, my measurement table is called T_Dach in Influx, that is with an underscore. If we stick to the standard I cant send an underscore ...

psy0rz, 2017-09-13:
If its part of the standard then this is correct behaviour. 

Can this be a bug in influxdb, since it doesnt properly decodes?

closed 2017-09-13 by psy0rz