letscontrolit/ESPEasy #463: PiDome MQTT protocol

Timm27, 2017-08-01:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
Tell us what should happen?

As soon as you enter the protocol "PiDom MQTT" and the server IP under Controllers, the IDX field disappears !

How to identify a sensor without IDX?

### Actual behavior
Tell us what happens instead?

In previous versions you could still enter IDX / variable for each sensor.

### System configuration

Hardware:
WeMos D1 mini

Software or git version:
ESPEasy_v2.0.0-dev11

Grovkillen, 2017-08-01:
Does pidome use IDX?

Timm27, 2017-08-02:
Yes, in the earlier versions.
I have always transferred the measured values of the sensors with the help of "PiDome MQTT" and "IDX".
I gave each connected sensor a unique IDX / variable and so the controller (FHEM) could assign the sensor data.
With firmware 2.0+ the IDX field disappears as soon as you select "PiDome MQTT" as the controller.
It is unfortunately no longer possible to use the "PiDome MQTT" protocol !!!

psy0rz, 2017-08-02:
this behaviour has changed: now you can enter the idx field for every device i think? (instead of globally per controller)

Timm27, 2017-08-02:
Unfortunately I do not see yet an IDX field (Release v2.0.0-dev11)

Under "Config" I first enter a unit number (1-31).
-----------------------------
Under "Controllers" I carry "PiDom MQTT" with IP address of FHEM server (port: 1883) -> and then "Enabled"
Controller Subscribe: / Home / #
Controller Publish: / hooks / devices /**% id%** / SensorData /**% valname%**
----------------------------
Under "Devices" I choose e.g. the sensor BMP085/180, assign a name and then "Enabled"

"**Send to Controller** ❶" is activated.

Now I see but no IDX field!

Without an IDX number, the FHEM server can not find the sensor!

In the configuration of FHEM, for example:

Attr BMP180 subscribeReading_Temperature / hooks / devices / **11** / SensorData / **Temperature**

Here is IDX = 11

pljakobs, 2018-03-12:
this is just a change in the topic. ESPEasy does not use the variable %id% anymore, but has replaced ith with %unit%. Change your topic string from /hooks/devices/%id%/SensorData to /hooks/devices/%unit%/SensorData and things will work again

TD-er, 2018-03-12:
The %unit% is a reference to the unit number next to the name of the ESPeasy node/module.
It is (currently) also used in the DNS name of the ESPeasy module.

Apart from that there are "idx" values, which can be assigned to each task (aka. plugin/device).
The idx is usually only used on Domoticz, to connect each value-set per sensor to a measurement group in Domoticz.
That allows to have multiple sensors per ESPeasy node. (e.g. CO2 / Temp / Humidity / Dust/ etc.)