letscontrolit/ESPEasy #465: MH-Z19 CO2 sensor gives empty MQTT messages

TD-er, 2017-08-05:
I just installed the pre-compiled v2.0.0-dev11 version on a NodeMCU with a MH-Z19 and a BMP280 sensor connected.
Both appear to work fine.
The data from the BMP280 sensor can be sent via Domoticz MQTT and Domoticz HTTP.
But the data from the MH-Z19 can only be sent via Domoticz HTTP.
The sensor gives stable readings (S-value = 64):

`26034969 : MHZ19: PPM value: 701 Temp/S/U values: 24/64/11051.00`

The logs in Domoticz, receiving MQTT and HTTP messages of the same sensor:

```
2017-08-05 15:46:35.460 MQTT: Topic: domoticz/in, Message: {"idx":73}
2017-08-05 15:46:35.549 Python: Changed: ID: 73 Name: ESPeasy CO2, Type: 243, subType: 31, switchType: 0, s_value: 701.0000, n_value: 0, n_value_string: 701.0000, last_update_string: 2017-08-05 15:45:35
```

And looking to the MQTT messages from 2 of these ESPeasy sensors:

```
pi@raspberrypi:~ $ mosquitto_sub -h localhost -t domoticz/in
{"idx":71}
{"idx":73}
{"idx":72,"nvalue":0,"svalue":"28.49;0;0;1014.23;0"}
{"idx":74,"nvalue":0,"svalue":"23.76;0;0;1013.91;0"}
```

- idx 71 and 73 are the MH-Z19 sensors.
- idx 72 and 74 are the BMP-280 sensors

I tried on 2 NodeMCU devices, also reboot and disable of the BMP280 sensor. But the MQTT messages are empty for the MH-Z19.

infernix, 2017-08-06:
It should log all these actions to syslog if you configure that. Example:

```
Aug  6 22:47:17 ESP Unit: 2 : MHZ19: PPM value: 540 Temp/S/U values: 26/64/10379.00
Aug  6 22:47:17 ESP Unit: 2 : MQTT : espeasy/bedroom2/MHZ19/PPM 540
Aug  6 22:47:17 ESP Unit: 2 : MQTT : espeasy/bedroom2/MHZ19/Temperature 26
Aug  6 22:47:17 ESP Unit: 2 : MQTT : espeasy/bedroom2/MHZ19/U 10379
Aug  6 22:47:17 ESP Unit: 2 : MQTT : Topic: espeasy/bedroom2/MHZ19/PPM
Aug  6 22:47:17 ESP Unit: 2 : MQTT : Payload: 540
Aug  6 22:47:17 ESP Unit: 2 : MQTT : Topic: espeasy/bedroom2/MHZ19/Temperature
Aug  6 22:47:17 ESP Unit: 2 : MQTT : Payload: 26
Aug  6 22:47:17 ESP Unit: 2 : MQTT : Topic: espeasy/bedroom2/MHZ19/U
Aug  6 22:47:17 ESP Unit: 2 : MQTT : Payload: 10379
Aug  6 22:47:17 ESP Unit: 1 : MHZ19: PPM value: 404 Temp/S/U values: 22/64/10368.00
Aug  6 22:47:17 ESP Unit: 1 : MQTT : espeasy/bedroom1/MHZ19/PPM 404
Aug  6 22:47:17 ESP Unit: 1 : MQTT : espeasy/bedroom1/MHZ19/Temperature 22
Aug  6 22:47:17 ESP Unit: 1 : MQTT : espeasy/bedroom1/MHZ19/U 10368
Aug  6 22:47:17 ESP Unit: 1 : MQTT : Topic: espeasy/bedroom1/MHZ19/PPM
Aug  6 22:47:17 ESP Unit: 1 : MQTT : Payload: 404
Aug  6 22:47:17 ESP Unit: 1 : WD   : Uptime 2232 ConnectFailures 0 FreeMem 21128
Aug  6 22:47:17 ESP Unit: 1 : MQTT : Topic: espeasy/bedroom1/MHZ19/Temperature
Aug  6 22:47:17 ESP Unit: 1 : MQTT : Payload: 22
Aug  6 22:47:17 ESP Unit: 1 : MQTT : Topic: espeasy/bedroom1/MHZ19/U
Aug  6 22:47:17 ESP Unit: 1 : MQTT : Payload: 10368
```

Are you seeing that?

TD-er, 2017-08-06:
The logs of ESP Easy, or logs in Domoticz?
I don't see these in the ESP easy logs.
In the Domoticz logs I see:

```
2017-08-06 23:15:53.222 MQTT: Topic: domoticz/in, Message: {"idx":73}
2017-08-06 23:15:53.223 Error: MQTT: Invalid data received!
2017-08-06 23:15:53.270 (Dummy) General/Custom Sensor (ESPeasy CO2)
2017-08-06 23:15:53.279 Python: Changed: ID: 73 Name: ESPeasy CO2, Type: 243, subType: 31, switchType: 0, s_value: 840.0000, n_value: 0, n_value_string: 840.0000, last_update_string: 2017-08-06 23:14:53
2017-08-06 23:15:53.279 Python: Number of user_variables: 0
```
The line starting with "Python" is the HTTP request.

infernix, 2017-08-06:
You need to run a syslog server and configure ESPEasy to send its syslog messages there.

TD-er, 2017-08-06:
I will install that tomorrow evening, when I am home from work again.

TD-er, 2017-08-07:
I installed a rsyslog server on a machine and started to receive the files in a logfile.

Just a grep on the lines with data from ESP sensor 1:
```
Aug  7 22:49:45 ESP Unit: 1 : MHZ19: PPM value: 1258 Temp/S/U values: 27/64/10395.00
Aug  7 22:49:45 ESP Unit: 1 : MQTT : {"idx":71}
Aug  7 22:49:45 ESP Unit: 1 : WD   : Uptime 101 ConnectFailures 0 FreeMem 19440
Aug  7 22:49:54 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 22:49:54 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 38,#012   "meterType" : "Energy",#012   "name" : "Zonnepanelen",#012   "nvalue" : 0,#012   "stype" : "YouLess counte
Aug  7 22:50:04 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 22:50:04 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 38,#012   "meterType" : "Energy",#012   "name" : "Zonnepanelen",#012   "nvalue" : 0,#012   "stype" : "YouLess counte
Aug  7 22:50:13 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 22:50:13 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "General",#012   "id" : "00082073",#012   "idx" : 73,#012   "meterType" : "Energy",#012   "name" : "ESPeasy CO2",#012   "nvalue" : 0,#012   "stype" : "Custom Sensor",
Aug  7 22:50:14 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 22:50:14 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 38,#012   "meterType" : "Energy",#012   "name" : "Zonnepanelen",#012   "nvalue" : 0,#012   "stype" : "YouLess counte
Aug  7 22:50:15 ESP Unit: 1 : WD   : Uptime 102 ConnectFailures 0 FreeMem 19440
Aug  7 22:50:17 ESP Unit: 1 : BMP280  : Address: 0x76
Aug  7 22:50:17 ESP Unit: 1 : BMP280  : Temperature: 30.35
Aug  7 22:50:17 ESP Unit: 1 : BMP280  : Barometric Pressure: 1018.91
Aug  7 22:50:18 ESP Unit: 1 : MQTT : {"idx":72,"nvalue":0,"svalue":"24.35;0;0;1018.91;0"}
Aug  7 22:50:20 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 22:50:20 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 100,#012   "RSSI" : 6,#012   "description" : "",#012   "dtype" : "Temp + Humidity",#012   "id" : "19214",#012   "idx" : 1,#012   "meterType" : "Energy",#012   "name" : "Zolder",#012   "nvalue" : 0,#012   "stype" : "Cresta, TFA TS34C
Aug  7 22:50:23 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 22:50:23 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 100,#012   "RSSI" : 7,#012   "description" : "",#012   "dtype" : "Temp + Humidity",#012   "id" : "27662",#012   "idx" : 58,#012   "meterType" : "Energy",#012   "name" : "Tuin2",#012   "nvalue" : 0,#012   "stype" : "Cresta, TFA TS34C
Aug  7 22:50:24 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 22:50:24 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 38,#012   "meterType" : "Energy",#012   "name" : "Zonnepanelen",#012   "nvalue" : 0,#012   "stype" : "YouLess counte
Aug  7 22:50:34 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 22:50:34 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 38,#012   "meterType" : "Energy",#012   "name" : "Zonnepanelen",#012   "nvalue" : 0,#012   "stype" : "YouLess counte
Aug  7 22:50:44 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 22:50:44 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 38,#012   "meterType" : "Energy",#012   "name" : "Zonnepanelen",#012   "nvalue" : 0,#012   "stype" : "YouLess counte
Aug  7 22:50:45 ESP Unit: 1 : MHZ19: PPM value: 1263 Temp/S/U values: 27/64/10395.00
Aug  7 22:50:45 ESP Unit: 1 : MQTT : {"idx":71}
Aug  7 22:50:45 ESP Unit: 1 : WD   : Uptime 102 ConnectFailures 0 FreeMem 19440
```

It is a bit more than expected (HTTP-sending data is switched off now for testing), but I see messages about the YouLess meter and some Cresta sensors also. But I never told Domoticz where the logserver was. Can it be that Domoticz found the rsyslog server via the ESP sensors?
Anyway, the MQTT messages to idx 71 are empty.

infernix, 2017-08-07:
Hmm. What if you set another mqtt type? Openhab for instance?

TD-er, 2017-08-07:
With OpenHab MQTT, I see the following in the log for that sensor:

```
Aug  7 23:28:48 ESP Unit: 1 : WD   : Uptime 12 ConnectFailures 0 FreeMem 19256
Aug  7 23:28:52 ESP Unit: 1 : BMP280  : Address: 0x76
Aug  7 23:28:52 ESP Unit: 1 : BMP280  : Temperature: 30.85
Aug  7 23:28:52 ESP Unit: 1 : BMP280  : Barometric Pressure: 1018.37
Aug  7 23:28:52 ESP Unit: 1 : MQTT : /Slaapkamer/bmp280/Temperature 24.85
Aug  7 23:28:52 ESP Unit: 1 : MQTT : /Slaapkamer/bmp280/Pressure 1018.37
Aug  7 23:28:56 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 23:28:56 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 38,#012   "meterType" : "Energy",#012   "name" : "Zonnepanelen",#012   "nvalue" : 0,#012   "stype" : "YouLess counte
Aug  7 23:29:06 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 23:29:06 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 38,#012   "meterType" : "Energy",#012   "name" : "Zonnepanelen",#012   "nvalue" : 0,#012   "stype" : "YouLess counte
Aug  7 23:29:16 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 23:29:16 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "General",#012   "id" : "00082073",#012   "idx" : 73,#012   "meterType" : "Energy",#012   "name" : "ESPeasy CO2",#012   "nvalue" : 0,#012   "stype" : "Custom Sensor",
Aug  7 23:29:16 ESP Unit: 1 : MQTT : Topic: domoticz/out
Aug  7 23:29:16 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 38,#012   "meterType" : "Energy",#012   "name" : "Zonnepanelen",#012   "nvalue" : 0,#012   "stype" : "YouLess counte
Aug  7 23:29:17 ESP Unit: 1 : MHZ19: PPM value: 400 Temp/S/U values: 27/64/10393.00
Aug  7 23:29:17 ESP Unit: 1 : MQTT : /Slaapkamer/co2/PPM 400.00
Aug  7 23:29:17 ESP Unit: 1 : MQTT : /Slaapkamer/co2/Temperature 27.00
Aug  7 23:29:17 ESP Unit: 1 : MQTT : /Slaapkamer/co2/U 10393.00
Aug  7 23:29:18 ESP Unit: 1 : WD   : Uptime 13 ConnectFailures 0 FreeMem 19256
```

infernix, 2017-08-07:
OK, the problem is that there is no support for SENSOR_TYPE_TRIPLE in Domoticz MQTT plugin, see [here](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_C002.ino#L146) - and MHZ19 is `Device[deviceCount].VType = SENSOR_TYPE_TRIPLE;`.

So the issue is not with the sensor but with the Domoticz MQTT code. A simple fix would be to change `_C002.ino` and copy/paste `SENSOR_TYPE_TEMP_HUM_BARO` and call it `SENSOR_TYPE_TRIPLE` but it might need some testing, and since I don't run Domoticz I wouldn't be able to.

Grovkillen, 2017-08-10:
A intermediate solution is to create the MQTT message with rules, see this tutorial:

https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules#Custom_reports_to_Domoticz_with_own_IDX

TD-er, 2017-08-10:
I just made a build with commit #468 and tested with Domoticz.
The outputs are as can be expected when looking at the code:
`Aug 11 00:33:35 ESP Unit: 1 : MQTT : {"idx":78,"nvalue":0,"svalue":"964.00;26.00;11603.00"}`

Domoticz can now draw a graph based on the 'custom' sensor profile via MQTT messages.
I do not know yet how to make a graph in Domoticz combining the PPM and the U values, but I guess that's just a matter of adding templates in Domoticz and is not something for ESPeasy.

I did not yet have a look at the output of the AirSense code. If that outputs extra values, it may be nice to base both outputs so the same template could be used for both sensors in Domoticz, OpenHab, etc. Although the U value of the MH-Z19 is rather device specific.

psy0rz, 2017-09-13:
thanks guys!

PR #468 (2017-09-13): Add SENSOR_TYPE_TRIPLE support to Domoticz MQTT
This should fix #465
closed 2017-09-13 by psy0rz