letscontrolit/ESPEasy #467: WiFi led stays ON after diable it and not possible to use GPIO

papperone, 2017-08-07:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Setup any PIN as WiFi Status LED, then disable it

Does the problem presist after powering off and on? NO

### Expected behavior
After disabling the WiFi Status LED from HW panel I am expecting it wiill be turned off and GPIO can be used as usual

### Actual behavior
After disabling WiFi Status LED the LED stays on and GPIO become unresponsive (e.g. GPIO,2,0 or GPIO,2,1 does not chage GPIO status)

### System configuration

Hardware: Wemos D1 mini

Software or git version: 2.0.0.dev11 (tested with dev10 and same behaviour)

psy0rz, 2017-09-13:
technically this is a bug, but normally pins are not "reset" to some default state after you unassign them.

as a workaround, just reset the unit.

closed 2017-09-13 by psy0rz

Grovkillen, 2017-09-13:
Or set the gpio back to "normal" using PWM command PWM,2,0